fluidd-core/fluidd #1150: Webcam does not show in multi instance Fluidd

AxMod3DPrint, 2023-08-14:
### Fluidd Version

1.25.1

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I have a Pi that only runs Fluidd, all of my printers are added to this instance via the top right 3 dots > add printer

All the standard webcams are working without issue, I have 1 printer that runs a Pi Camera this does not show in this instance. All the printers have a Fluidd install via KIAUH as a quick method to get NGINX set up. When I go to it locally http://<printer>.local the webcam shows. 

Due to using a centralised Fluidd instance, I have to use `http://<printer>.local/webcam/?action=stream` (`/snapshot`) this does not work on the centralised instance but does on the local instance, on the local instance `webcam/?action=stream` also works without issue. I also have to use Spyglass over Crowsnest as I do not get an image with Crowsnest and this camera. 

I can also view this camera directly via the URL, so I don't think it's spyglass that would have the issue.

Possibly linked to #1142 - but I'm not too sure, could be just a niche issue.

### What did you expect to happen

Webcam to show in the Centralised instance of Fluidd

### How to reproduce

Add more than 1 printer to Fluidd, use a Picam and spyglass - set up using `http://<printer>.local/webcam/?action=stream - cam will not show

### Additional information

Centralised Fluidd - Cam does not show
[image]

Local Fluidd - cam shows
[image]

Browser View - Cam shows
[image]

Settings
[image]

recrudesce, 2023-08-14:
This is a CORS issue that does not appear in Mainsail combined instances.

[image]

pedrolamas, 2023-08-15:
CORS is a security feature maintained by the server (moonraker or in this case, the streamers) and the browser - Fluidd has little to no control over CORS.

Have you tried without the ".local"? I ask this as ".local" is considered a Bonjour (mDNS) service indicator, not a proper domain - and in fact its use as local domain should be avoided!

recrudesce, 2023-08-15:
The issue doesn't exist with a central Mainsail instance when pointed at the same camera endpoints (I run both)

**Mainsail showing camera**
[image]

**Fluidd not showing camera**
[image]

CORS coming back from ustreamer etc is showing `Access-Control-Allow-Origin: *`

I have tried IP addresses, DNS names, mDNS names.  Nothing works in Fluidd but will work in Mainsail with absolutely no config changes.

AxMod3DPrint, 2023-08-15:
No, I'll test this later, however I do have a local DNS server that
provides the DNS to the static IP and isn't using Bonjour

pedrolamas, 2023-08-15:
> This is a CORS issue that does not appear in Mainsail combined instances.
> 
> [image]
> 
> EDIT: This happens regardless of whether you use Crowsnest, Spyglass, or a manual ustreamer instance. The issue presumably lies Fluidd's end.

If you right-click the above link on the image and select "open in new tab", does the stream work?

recrudesce, 2023-08-15:
> If you right-click the above link on the image and select "open in new tab", does the stream work?

Yes, cos by that point the browser goes straight to the camera server itself and isn't being requested by the Fluidd instance.

pedrolamas, 2023-08-15:
I may have found the issue: the link above says "?action=snapshot" but in Fluidd settings you seem to be using "MJPEG Stream", so I would expect this to be "?action=stream"

Can you double-check the URLs you have on the camera settings on the "Camera URL stream" and "Camera URL snapshot"?

recrudesce, 2023-08-15:
[image]

recrudesce, 2023-08-15:
Same settings as per central Mainsail instance
[image]

pedrolamas, 2023-08-15:
The settings are currently shared between Fluidd and Mainsail (both are using the "webcam" namespace in Moonraker)

A couple of things to try:
 - Ensure you are running the latest Fluidd 1.25.1
 - Clear browser caching

You got my attention on this issue! If you are around Discord, I'll be happy to discuss this directly on chat!

pedrolamas, 2023-08-15:
@matmen, could this somehow be related with the `crossorigin="anonymous"` that was added to the cameras on last release?

recrudesce, 2023-08-15:
I assume it's something to do with 1.25.1, as it worked fine in previous versions.  We're both running 1.25.1 (I'm specifically running your Docker container).

I'm in the Voron Discord as @recrudesce - feel free to prod me if you need any other info :)

pedrolamas, 2023-08-15:
I've confirmed that indeed this was a bug introduced in the latest version with the addition of `crossorigin="anonymous` to the various camera components.

We will quickly fix this and release in the upcoming v1.25.2

pedrolamas, 2023-08-15:
@recrudesce fix is now part of the "develop" branch, and as you are using Docker, I wonder if you are willing to give the dev container a test (use the ":latest-develop" docker image tag)?

recrudesce, 2023-08-15:
Yup, that fixed it !

[image]