fluidd-core/fluidd #1160: Wrong language selected

pmarkiewicz, 2023-08-28:
### Fluidd Version

1.25.2

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Language set in settings is english, but fluidd displays in OS language (in my case polish)

### What did you expect to happen

Respect language selected in settings not one from OS

### How to reproduce

Select language different than OS (or maybe browser default). Looks like this is limited to Cr as I can't reproduce on FF

pedrolamas, 2023-08-29:
Hi @pmarkiewicz thank you for reporting this.

We've seen other users having the same issue but have yet to determine what is causing this behavior.

Having said that, some users found that they could mitigate this problem by doing a full browser refresh of Fluidd (with <kbd>Ctrl</kbd>+<kbd>F5</kbd>), then going on Fluidd Settings, changing the language to English, and then back to their chosen language and it would then work!

Can you please check if these steps also fixes the problem for you?

pmarkiewicz, 2023-08-29:
Looks like your solution works, thx

closed 2023-09-01 by pedrolamas

Devsider, 2024-04-05:
I have often been experiencing the same issue. 
Even tho its not that big of a problem, its still annoying. 
Going into the settings and switching the language to something else and back to english seems to fix it - temporarely, until it happens again.