fluidd-core/fluidd #1161: NoScript warning when accessing via hostname, IP address works fine

ccclever2, 2023-08-30:
### Fluidd Version

v1.25.1-513523f

### Browser

Chrome, Firefox, Safari, Edge

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, macOS, Linux, Android

### What happened

I am experiencing a really odd bug. When i try to access my fluidd instance using its hostname (`fluidd`) i get a blank page. In the page source code i see the basic DOM and a noscript warning:

`` <strong>We're sorry, but fluidd doesn't work properly without JavaScript enabled. Please enable it to continue.</strong>
 ``

When i query the hostname on the same system `dig fluidd` it returns me a local IP address. When now using this IP address in the browser to access the web interface, everything works fine, i see the interface and no noscript warning. Because i still get a response from fluidd when using the hostname, i conclude that it cannot be a DNS issue. 
Any ideas?

### What did you expect to happen

Expected a working interface when using hostname `fluidd`

### How to reproduce

Installed fluidd using `kiauh`, query `fluidd` (chosen hostname) and the resolving IP in the browser

pedrolamas, 2023-09-07:
Hi @ccclever2, thank you for reporting this.

As you can access via IP, I doubt this is a bug on Fluidd, so I've updated the labels on this ticket.

This sounds more like some issue with some extension/anti-virus than anything else... can you press <kbd>F12</kbd> on the browser to bring up the Developer Tools and then check the Console tab there for any error message that might be logged?

pedrolamas, 2023-09-20:
Closing this now as there has been no more contact regarding this matter.

If you still need help with this, I would recommend that you try [Klipper Discord server](https://discord.klipper3d.org/) instead.

closed 2023-09-20 by pedrolamas