fluidd-core/fluidd #1169: Ability to reset Thermals Graph to clear erroneous RTD Amplifier temperatures during Amplifier initialization

ChaosBlades, 2023-09-08:
### Is your feature request related to a problem? Please describe

When you use an RTD Amplifier such as the MAX31865 on the BTT EBB36 you will get erroneous temperature readout during that chips initialization. This behavior can be reproduced simply by initiating a FIRMWARE_RESTART Would be nice if you could avoid the Thermals Graph becoming useless as these readings re-scale the graph to uselessness.

### Describe the solution you'd like

Ability to clear the graph information without having to reboot the printer.

### Describe alternatives you've considered

Ability to set scaling limits for the graph.

### Additional information

[image]
[image]

pedrolamas, 2023-09-08:
Hi @ChaosBlades thank you for reporting this.

I agree, this is a really bad experience (though it is one outside of Fluidd control), but I have to admit I am a bit skeptical in doing such a fix in Fluidd as it feels a bit that this should be done upstream in Moonraker or even Klipper!

I will discuss this with the other devs and see what we can do to improve this.

ChaosBlades, 2024-04-21:
@pedrolamas Looks like this isn't going to happen upstream. Moonraker wants Klipper to fix this. I posted to the Klipper forum and got no useful responses to this issue.

https://klipper.discourse.group/t/erroneous-rtd-amplifier-temperatures-during-amplifier-initialization-when-save-config/15022/2

Can we just add an option in Settings to limit the min/max scaling values in the graph? So it scales like normal up to the values defined. So I would probably set 300C Max and -10C Min.

pedrolamas, 2026-10-15:
As per discussion in https://github.com/Klipper3d/klipper/pull/7406, the latest changes in Klipper (specifically, https://github.com/Klipper3d/klipper/pull/7212) seem to mitigate this issue, so I am closing this down now.

Nevertheless, and as discussed, if this issue does resurface, I do believe it should be fixed upstream in Klipper instead of anything we do here in Fluidd.

closed 2026-10-15 by pedrolamas