fluidd-core/fluidd #1173: Option to disable filament type check

charredchar, 2023-09-13:
### Is your feature request related to a problem? Please describe

I am using a multi-material unit and Fluidd will warn that the material selected does not match the filament type from the gcode during the start of a print even when it does. In my specific case I will slice a file with only ASA, the UI shows ASA under type (mind you, multiple times depending on how many filaments I have selected for that gcode) but it'll still warn me it doesn't match when I select a spool of ASA for Spoolman.

### Describe the solution you'd like

Personally I think just including an toggle in the settings to disable this is fine for me as I won't ever mix it up.

### Describe alternatives you've considered

Fluidd could also be set to check just the first filament type against the spool selected at a print start as you may want a different filament type later. An example would be for support materials being different than the base print material.

pedrolamas, 2023-09-21:
Hi @charredchar, thanks for reporting this.

Can I ask that you upload a gcode sample file and a screenshot of Fluidd showing your selected spool where you have experienced this?

charredchar, 2023-09-21:
Sure thing. It'll give me the error regardless of what materials are actually printing because I have 8 extruders listed in PrusaSlicer and it seems to include them all in the Filament type no matter what.
(I have to rename it to `.txt` to upload)
[Printer Standoff.txt](https://github.com/fluidd-core/fluidd/files/12693322/Printer.Standoff.txt)
[image]
[image]
[image]

pedrolamas, 2023-09-21:
Well, I guess you answered the question yourself with that: we are comparing "ASA" with "ASA;ASA;ASA;ASA;ASA;ASA;ASA;ASA" and obviously, that is different! 🙂

I have some doubts if we should even be making this check as it can be quite erroneous... some users don't change the filament type on the slicer, they just change temperatures and call it a day!

@matmen any thoughts on this?

charredchar, 2023-09-21:
> Well, I guess you answered the question yourself with that: we are comparing "ASA" with "ASA;ASA;ASA;ASA;ASA;ASA;ASA;ASA" and obviously, that is different! 🙂
> 
Understand the reason it happens but I have no way to stop it since it is something that comes directly from the slicer.
The check itself isn't a terrible idea... but letting the user choose if they want it or even doing the alternative to (also) have a option to only check just the first filament, ie ignore everything after the first `;` in the filament type.
I haven't done any tests myself though to see if just checking the first filament listed would work. I don't know if the first filament is always T0 from the slicer or is actually the first filament that gets printed regardless of which tool it is. I really only ever print in ASA with the rare exception being some fancy PLA.

matmen, 2023-09-22:
@pedrolamas I'm not sure how consistent the slicers are here (separation characters, quotes around filament names), but I think we're fine if we just check for something like `print_filament_type.split(';').includes(spool_filament_type)`?
I don't think we can reliably check if the selected spool is the one actually needed for the section of Gcode, but it would at least fix this issue.
Having a toggle for the sanity checks would be good too.

I'll take a look at this over the weekend :)

pedrolamas, 2023-09-23:
Well, this is the kind of stuff I've seen from slicers:

- PLA
- PLA;PLA
- "PLA;PLA"
- "PLA";"PLA"

A couple of months ago I did a PR to Moonraker to trim the start and end double quotes (which was merged), so there is also this:

- PLA";"PLA

so split, trim quotes, and then check the result array.

There are also the + variants (like PLA+), where the slicer just has PLA but user set spool to "PLA+" or even "PLA Plus" or something like that....

I would seriously consider having a toggle setting for this check!

charredchar, 2023-09-23:
> There are also the + variants (like PLA+), where the slicer just has PLA but user set spool to "PLA+" or even "PLA Plus" or something like that....

Oh this is a really good point. Maybe I'll post a request to SpoolMan to turn the filament type field into a predefined drop down so people can't just enter "PLA Plus" instead of "PLA+". It's more restrictive to end users but will stop those edge cases.

pedrolamas, 2023-09-23:
@charredchar though I like the sentiment, it will not fix the issue as users can then write whatever they want on the slicers... Cura for example has a text field, so users can write whatever they want!

charredchar, 2023-09-23:
> @charredchar though I like the sentiment, it will not fix the issue as users can then write whatever they want on the slicers... Cura for example has a text field, so users can write whatever they want!

Ah, true. I was thinking of all the filament presets available that already have the type populated.

PR #1186 (2023-09-23): feat(spoolman): allow toggling individual sanity check warnings
Adds two new Spoolman settings:
[image]

Fixes #1173
closed 2023-09-23 by matmen