fluidd-core/fluidd #1178: Spoolman Dialog Shows Wrong Times

Ocraftyone, 2023-09-18:
### Fluidd Version

v1.25.3-26d94c8

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

The spoolman dialog does not take into account timezones. I have both the timezone on my pi and the timezone in the spoolman docker container set to the same zone. Spoolman seems to store the times in UTC then converts the time on the fly when displaying to the user (as evident by when you edit the spool it is in UTC and when just viewing the time it is converted to the proper timezone). Fluidd just needs to take into account the timezone before displaying when it was last used.

Fluidd Spoolman Dialog:
[image]

Native Spoolman Page:
[image]

Under Edit Spool in native spoolman page:
[image]

### What did you expect to happen

The timezone is taken into account when calculating when the spool was last used.

### How to reproduce

Use a spoolman instance and fluidd instance with timezones

pedrolamas, 2023-09-19:
Hi @Ocraftyone thank you for reporting this.

From a quick check on Fluidd code, I believe we are using UTC dates and assuming that is what Spoolman API is giving us - so that will be the first thing to check.

We will report back when we know more.

pedrolamas, 2023-09-20:
I have now confirmed and reproduced this issue, though I would argue this is a Spoolman API issue, not a Fluidd one!

As you well said and I confirmed in the Spoolman docs, the Spoolman API returns dates & times in UTC.

However, the string representation of said dates & times is missing the time zone indicator as per [ISO 8601](https://en.wikipedia.org/wiki/ISO_8601).

For example, calling `/api/v1/spool`, we get a list of spools and I can see the data in all date fields have no time zone indication:

[image]

As these are all UTC dates & times, these strings should end with a single "Z" - following the ISO and thus solving this issue in Fluidd and any other frontend.

I will open an issue with Spoolman regarding this and see if they want to fix it upstream as I believe it should be.

pedrolamas, 2023-09-20:
This issue has now been fixed upstream in https://github.com/Donkie/Spoolman/commit/b795680226b8ba8f93b4895bcaae379acc176242

closed 2023-09-20 by pedrolamas