fluidd-core/fluidd #1203: When a URL including a path is specified, fluidd dispatches requests to / anyway.

Laikulo, 2023-10-08:
### Fluidd Version

v1.25.2-e06c7ad

### Browser

Firefox, Other (Please tell us below)

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, Linux, Android

### What happened

When a moonraker url in the form `http://printhost.example.net/printer1` is provided, the URL is accepted, but requests to the moonraker API are dispatched to `http://printhost.example.net/ACTION`

### What did you expect to happen

Requests to be dispatched to `http://printerhost.example.net/printer1/ACTION`

### How to reproduce

Given an print host with hostname printhost.example.net:
With moonraker ae1d3b039308a84cbc01092f11b4c7c3b84da138 or later serving on localhost:8080, configure route_prefix to "moonraker1"
Configure an httpd listening on 80 that reverse proxies /moonraker1 to localhost:8080, with websockets enabeled.
Configure the httpd to serve fluidd's htdocs for other requests.

Navgate to `http://printhost.example.net`
In the "add printer" dialogue that appears, enter `http://printhost.example.net/moonraker`

pedrolamas, 2023-10-08:
HI @Laikulo thanks for opening this ticket.

The "route_prefix" setting is a recent Moonraker feature, hence why we do not support it yet. As such I've changed this ticket classification from Bug to Feature Request.

We will look into this as soon as possible!

closed 2023-10-09 by pedrolamas

Laikulo, 2023-10-10:
Can confirm that this now works as expected!

PR #1205 (2023-10-09): feat: adds moonraker "route_prefix" support
Adds support for Moonraker new `route_prefix` setting, where a user can specify a subpath for the server such as "moonraker/printer1"

Resolves #1203