fluidd-core/fluidd #1210: Add to fluidd config to ignore auto_save prompt for bed_tilt_calibrate like bed_mesh is currently

DaVinci-10, 2023-10-21:
### Is your feature request related to a problem? Please describe

Would it possible to add the bed_tilt_calibrate to a list of save_config ignore messages like
`Ignore default bed mesh Pending Configuration Changes`

As I have it run before each print, It would be helpful to ignore the prompt to save like bed_mesh.
[image]

### Describe the solution you'd like

Add another slider or have the same one apply to both
[image]

### Describe alternatives you've considered

Right now I just restart to not save.

pedrolamas, 2023-10-23:
Hi @DaVinci-10, thanks for opening this Feature Request.

I see value in this, and we will indeed implement this.

Having said that, I am considering if we should instead add a customizable area to specify sections to ignore, instead of just adding yet another toggle switch...

It would look something like this:

[image]

@matmen, any thoughts on this?

matmen, 2023-10-24:
@pedrolamas I think that's a good idea, not sure if it should be a multi-select with a predefined list of sections or if it should be freely configurable by the user - but I guess the latter provides more flexibility

PR #1217 (2023-10-26): feat: ignore pending config changes in custom sections
Replaces the existing "Ignore default bed mesh Pending Configuration Changes" boolean setting with a new configurable "Sections to ignore Pending Configuration Changes" combo setting.

It defaults to an empty list, but when clicked it will provide two items that can be easily added by clicking on them.

[image]

Besides those items, users will be able to add whatever they want to the list.

Any pending configuration changes on these sections will just be ignored and will NOT cause the "Save Config & Restart" button to show (unless some oth [... cut]
closed 2023-10-26 by pedrolamas