fluidd-core/fluidd #1212: Add quantity field to print queue items

Ivanuke, 2023-10-22:
### Is your feature request related to a problem? Please describe

I mass produce parts on my print farm that utilize continuous printing feature. The "Add To Queue" option is great but in order to print hundreds of copies of the same file, I have to click "Add to Queue" the same number of times. This is a very inefficient way to build up the print queue when requiring many multiples of a file to be printed repeatedly for extended periods of time.

### Describe the solution you'd like

Is it possible to add a quantity field, next to each unique file to be looped through before moving onto the next unique file? For Example

File A - Quanity: 75
File B - Quantity 75
File C - Quantity 75

The following would currently require me to press "Add to Queue" 225 times, manually keeping count.

### Describe alternatives you've considered

No alternatives, currently adding files manually one by one to the Jobs queue.

pedrolamas, 2023-10-23:
Hi @Ivanuke, thank you for opening this Feature Request.

At first glance, I see two options here:

1. the current "add to queue" context menu item would show a dialog asking how many copies to add
2. we would add a new context menu item to the queued item with a "multiply" that would then show a dialog asking for how many copies to create in the queue (on the same position as the current one)

Though we can do both here, I am inclined on implementing only option 2 above

Ivanuke, 2023-10-23:
Option 2 would be great! Excited to have this feature and hopefully it can be done!

PR #1213 (2023-10-24): feat: adds Multiply context menu item to Job Queue
Adds a new Multiply context menu item (and bulk item selection toolbar action) that allows to enqueue multiple copies of the same queued job.

## Context menu item

[image]

## Bulk selection toolbar action

[image]

## Multiply Job dialog

[image]

Resolves #1212