fluidd-core/fluidd #1214: [FR] Ability to expose native Klipper commands as buttons

Sineos, 2023-10-24:
### Is your feature request related to a problem? Please describe

Just stumbled over this example: https://klipper.discourse.group/t/internal-error-bed-screws-adjust/11159/2

I find it horrid to create a macro just to be able to display built-in Klipper commands as a button in the UI.

### Describe the solution you'd like

Provide means to map built-in Klipper commands to buttons.

pedrolamas, 2023-10-26:
Hi @Sineos, thank you for sending this Feature Request.

I do see value in this request, though I haven't yet given proper thought on how we would go to implement this...

Nevertheless, for the specific case that the link refers, we already expose all of those directly on the Fluidd interface!

The `BED_SCREWS_ADJUST` is on the "Tools" menu in the Tool card:

[image]

Once the user clicks that, the "Bed Screws Adjust" dialog will pop up with the tool status and action buttons:

[image]

The dialog will only show if this option is enabled though:

[image]

Sineos, 2023-10-26:
Thanks for coming back to this @pedrolamas.

Yes, I'm aware of the above functionality and the initial example was not the best. I just have seen such "button macros" quite sometimes in support activities and finally thought, I just place this FR, since I think it is cluttering and "abusing" the `printer.cfg` for an UI inherent functionality.

Maybe a simplified version of Octoprint's [Custom Controls](https://docs.octoprint.org/en/master/features/custom_controls.html), e.g. just a control name and an underlying command.

pedrolamas, 2023-10-26:
I wasn't aware of the OctoPrint Custom Controls, but that sure has spiked my interest!