fluidd-core/fluidd #1234: Broken mesh

digitalninja-ro, 2023-11-11:
### Fluidd Version

1.26.2

### Browser

Firefox, Safari

### Device

Desktop PC, Laptop

### Operating System

macOS

### What happened

The mesh is a mess. But only on fluidd. I have also installed MainSail  on the same pi3 and everything seem to work perfect. 

Same mash on:
Fluidd
[image]
MainSail
[image]

This is happen for me with the last version of fluidd 1.26.2
[image]

### What did you expect to happen

Smooth mesh

### How to reproduce

I don't know how to reproduce.

pedrolamas, 2023-11-11:
Hi @digitalninja-ro, thank you for opening this issue.

Can I ask that you please upload a klippy.log from the printer where this is happening?

digitalninja-ro, 2023-11-11:
Hi @pedrolamas thank you for the quick answer. Here is my klippy.log
[klippy.log](https://github.com/fluidd-core/fluidd/files/13325064/klippy.log)

PR #1235 (2023-11-11): fix: corrects order of e-charts dimensions
Corrects the order or e-charts dimensions.

Tested with sample config:

```ini
[bed_mesh]
speed = 400
mesh_min = 5,5
mesh_max = 225,182
probe_count = 5, 4

[bed_mesh default]
version = 1
points = 
  -0.016250, -0.010833, 0.001667, 0.004167, 0.047500
  0.035000, 0.023750, 0.018750, -0.015000, 0.046667
  0.020417, 0.002083, 0.027500, 0.007500, 0.039167
  -0.000833, -0.030000, -0.028333, -0.019583, 0.008333
tension = 0.2
min_x = 5.0
algo = lagrange
y_count = 4
mesh_y_pps = 2
min_y = 5.0
x_count = 5
max_y = 182.0
mesh_x_pps = 2
max_x = 225.0
```

Results: [... cut]
closed 2023-11-11 by pedrolamas