fluidd-core/fluidd #1237: Voron logo in top left corner became black instead of red

sprior, 2023-11-15:
### Fluidd Version

v1.26.3-79daaf0

### Browser

Firefox

### Device

Laptop

### Operating System

Windows

### What happened

I just noticed that the Voron logo in the top left has changed from red to black.

### What did you expect to happen

red Voron logo

### How to reproduce

select the Voron theme

pedrolamas, 2023-11-15:
Hi @sprior thank you for opening this issue, can you please check if doing a refresh with Ctrl+F5 or clearing the browser cache fixes the problem?

sprior, 2023-11-15:
@pedrolamas using FIrefox. Ctrl+F5 didn't change it but clear browser cache did.  I wonder how that happened.

closed 2023-11-15 by sprior

pedrolamas, 2023-11-15:
The vendor logos were all updated on the last release, I assume the browser was still using cached logos instead of the updated ones, and so this issue surfaced.

Nevertheless, we at least know that clearing the cache fixes the issue!

dpremo-ul, 2024-02-11:
I got my printer in January 2024.  For me, the Fluidd logo in the top left corner is black.  

If I open an incognito window, the Fluidd logo is blue, or matches the theme color I choose.  In a normal browser window, the Fluidd logo stays black even if I choose a different color.  Other logos update to reflect my color choice.  

Clearing cache or deleting browser data doesn't resolve this.  I only clear data for this site.  It's annoying, but it would be more annoying to delete cookies for all sites.  

I have this problem on multiple machines.  Only the Fluidd logo.

pedrolamas, 2024-02-11:
@dpremo-ul can I ask what browser and OS are you using?

For what it's worth, the fact that it works on an anonymous window leaves me thinking that somehow the cache is not being completely cleared - either that or we have an extra caching level outside of the browser!