fluidd-core/fluidd #1238: After update: Value at option 'repo: cadriel/fluidd' does not match detected repo 'fluidd-core/fluidd', falling back to detected version.

dmit2k, 2023-11-17:
### Fluidd Version

v1.26.3

### Browser

Firefox

### Device

Laptop

### Operating System

macOS

### What happened

I have Fluidd installation on a Raspberry Pi3 which worked fine last 2 years with periodic updates. I haven't used nor updated it for last several months but yesterday initiated the update process from the GUI. It updated and seems to work well but now shows me the following notice on the Update page:

 fluidd
v1.26.3
UP TO DATE
(i) Value at option 'repo: cadriel/fluidd' does not match detected repo 'fluidd-core/fluidd', falling back to detected version. 

### What did you expect to happen

Can someone please explain what could be the reason and how can I fix it?

### How to reproduce

Unfortunately I don't remember the Fluidd version I had before starting update

pedrolamas, 2023-11-17:
Hi @dmit2k, thank you for opening this issue, can you please upload your latest moonraker.log so we can take a look?

dmit2k, 2023-11-17:
@pedrolamas Thanks for the reply, here it is

[moonraker.log](https://github.com/fluidd-core/fluidd/files/13395059/moonraker.log)

pedrolamas, 2023-11-17:
The issue seems to be in your moonraker.conf, just replace "cadriel/fluidd" with "fluidd-core/fluidd", restart Moonraker, and that should solve the problem.

firsttest12, 2023-11-17:
tx. FWIW I had the same issue and was "only" updating from 1.26.1

pedrolamas, 2023-11-17:
I believe that message was just a warning from Moonraker, it now performs those sanity checks.

If the issue is fixed, please close this ticket.

dmit2k, 2023-11-17:
Thank you, that worked fine!

closed 2023-11-17 by dmit2k

buikpijn, 2023-12-13:
Thank you, this solved it.