fluidd-core/fluidd #1241: Filament required is unknown when not using Extruder 0.

charredchar, 2023-11-17:
### Fluidd Version

v1.26.2-968bdc8

### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

When slicing a file on a printer with multiple extruders (using an MMU, etc) and you don't have the first or "default" extruder printing anything Fluidd will warn that it does not know how much filament is needed for the print.

### What did you expect to happen

Fluidd checks either the first filament used or total used and compare it to the selected spool.

### How to reproduce

Slice a using multiple extruders but do not have anything using the first or "default" extruder.

### Additional information

Digging deeper into this I *think* Fluidd is checking the `filament used` fields in the gcode file. A normal file with multiple extruders but only using the first extruder will look like this.

```
; filament used [mm] = 35399.09, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00
; filament used [cm3] = 85.14, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00
; filament used [g] = 95.36, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00
; filament cost = 2.86, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00
; total filament used [g] = 95.36
; total filament cost = 2.86
; estimated printing time (normal mode) = 3h 12m 39s
; estimated first layer printing time (normal mode) = 2m 47s

```
Slicing with only using the 5th extruder will look like this.

```
; filament used [mm] = 0.00, 0.00, 0.00, 0.00, 24312.35, 0.00, 0.00, 0.00
; filament used [cm3] = 0.00, 0.00, 0.00, 0.00, 58.48, 0.00, 0.00, 0.00
; filament used [g] = 0.00, 0.00, 0.00, 0.00, 65.50, 0.00, 0.00, 0.00
; filament cost = 0.00, 0.00, 0.00, 0.00, 1.96, 0.00, 0.00, 0.00
; total filament used [g] = 65.50
; total filament cost = 1.96
; estimated printing time (normal mode) = 2h 55m 27s
; estimated first layer printing time (normal mode) = 1m 57s
```
As you can see the first few are all zeros because I never used those extruders. It would be simple to just use the `total filament used` field instead but this would also be inaccurate for selecting a spool at start when you're using say 3 different spools during the entire print. I have no clear answer on how to resolve this beyond making some attempt to make each comma separated filament used to a spool in Spoolman at the start of every print... Or reach the GCode file for how much filament is used between the start of a print and the switch to a new tool and only use that for the initial print selection.
But so far if I am not using only a single spool I basically have to completely ignore the first selection and its warnings when starting a print.

pedrolamas, 2023-11-17:
Hi @charredchar, thanks for opening this issue.

Can I ask that you upload a sample gcode file where you see this happening so we can use it for testing?

charredchar, 2023-11-21:
Sorry for the delay, I wanted to slice files that were exactly the same except for which extruder is selected and verify before uploading. The one labeled with `T5` will return the `Filament amount is unknown` error.
[Oneplus_6_Light.zip](https://github.com/fluidd-core/fluidd/files/13431074/Oneplus_6_Light.zip)

pedrolamas, 2023-12-01:
Thank you for providing the files @charredchar, I just tried this and can confirm the problem.

Having said this, this needs to be fixed upstream in Moonraker, as the "total_filament" is being returned from Moonraker API as 0.

closed 2024-06-12 by pedrolamas