fluidd-core/fluidd #1261: Tool change macro with active=True is not highlighting the active tool

viesturz, 2023-12-07:
### Fluidd Version

v1.26.3

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I'm trying to get the active tool button to have a different color, as per the screenshot in  https://github.com/fluidd-core/fluidd/pull/1111

### What did you expect to happen

The acttive tool should be highlighted.

### How to reproduce

Create Gcode macros for T0, T1, T2, etc.

```
[gcode_macro T0]
variable_active: True
gcode:
  SELECT_TOOL T=0
```

Expecting this to highlight the button. But it's the same as others.
Also tried adding the variable_color: "F00" but no effect.

The only option that works is variable_color: "FF0000"

viesturz, 2023-12-07:
Also colors for T0 and T1 are not working at all on one printer, but fine on another.
Some config corruption?

viesturz, 2023-12-07:
[image]

viesturz, 2023-12-07:
My macros are super simple 

``` 
[gcode_macro T0]
variable_color: "f00"
gcode:
  SELECT_TOOL T=0

[gcode_macro T1]
variable_color: "f00"
gcode:
  SELECT_TOOL T=1
````
Etc.

As I said, only T0 and T1 are not updating the color, and only one printer out of 2, so likely some snafu on my side.

pedrolamas, 2023-12-07:
@viesturz that looks ok to me... I would check if you have some hidden spaces on that which could cause this issue.

Having said that, Fluidd does not keep what tool is active, that is up to your macros to do (each `Tx` macro needs a `variable_active: False` or `True` to set the active state).

viesturz, 2023-12-07:
Yes I know that. Just was expecting that the active parameter is actally used, since it is populated in the code of #1111.

Digging into the web traffic, and this is wierd.
I changed the configs back to empty string colors (with a macro to override it on toolchange).
But for T0 and T1 the variables have active=. Which I was experimenting with some hours ago, but now they should be removed. WTF?

Request:
```
http://192.168.86.38/server/database/item?namespace=fluidd
```

Response:
```
"macros": {
                "stored": [
<snip>
              {
                        "name": "t0",
                        "alias": "",
                        "visible": false,
                        "disabledWhilePrinting": true,
                        "color": "",
                        "categoryId": "0",
                        "variables": {
                            "active": false
                        }
                    },
                    {
                        "name": "t1",
                        "alias": "",
                        "visible": false,
                        "disabledWhilePrinting": false,
                        "color": "",
                        "categoryId": "0",
                        "variables": {
                            "active": true
                        }
                    },
                    {
                        "name": "t2",
                        "alias": "",
                        "visible": false,
                        "disabledWhilePrinting": false,
                        "color": "",
                        "categoryId": "0",
                        "variables": {
                            "color": ""
                        }
                    },
                    {
                        "name": "t3",
                        "alias": "",
                        "visible": false,
                        "disabledWhilePrinting": false,
                        "color": "",
                        "categoryId": "0",
                        "variables": {
                            "color": ""
                        }
                    },
]
```

The printer objects status over the websocket is correct however. 
```
gcode_macro T0: {color: ""}
gcode_macro T1: {color: "c44"}
gcode_macro T2: {color: ""}
gcode_macro T3: {color: ""}
gcode_macro T4: {color: ""}
gcode_macro T5: {color: ""}
```

PR #1263 (2023-12-08): feat: highlight active tool
Adds support for the `Tx` macros `variable_active` so that the active tool is highlighted.

[image]

Resolves #1261
closed 2023-12-08 by pedrolamas