fluidd-core/fluidd #1270: K1 Max Camera modal collapsed, won't expand after recent update!

morriscode, 2023-12-11:
### Fluidd Version

1.26.3

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

After a recent update to Fluidd, the camera modal on the main page, presents the camera as normal, then after a few seconds it's collapses. There are several posts on reddit with videos and screenshots of this behavior also on K1 Max's.

https://www.reddit.com/r/klippers/comments/17vkgw7/camera_view_offsets_and_disappears_in_fluidd_only/

```
Camera settings have not changed and are as follows:
Enabled (toggled on)
Name: Camera
Flip Horizontally (default off)
Flip vertically (default off)
Rotate by: None
Stream type: MJPEG Adaptive
FPS Target: 30
FPS Target when not in focus: 5
Camera Url Stream: /webcam?action=stream
Camera Url Snapshot: /webcam?action=snapshot

```
I have tried MJPEG Stream with the same behavior. The camera works for a little while then collapses into this box.

[image]

I've removed and readded the camera several times with the behavior persisting. 
Previously have bene using Fluidd with this printer with no camera issues until I updated. 

The only apps installed are 
- fluidd
- klipper
- moonraker

Fwiw. The camera renders fine in crealty's slicer
|
[image]

### What did you expect to happen

The camera to continue working as it has been and not self collapse into a box rendering the camera view ineffective.

### How to reproduce

Connect to a Crealty K1 Max

morriscode, 2023-12-11:
I fixed this btw, after running packet captures, I noticed the GET request to the webcam was attempting to hit port 4408, it needs to attempt on port 4409. 

After replacing the values 
```
Camera Url Stream: /webcam?action=stream
Camera Url Snapshot: /webcam?action=snapshot
```
With the following full uri struct it works fine
```
Camera Url Stream: http://192.168.1.213:4409/webcam/?action=stream
Camera Url Snapshot: http://192.168.1.213:4409/webcam/?action=snapshot
```

Leaving open as I doubt it's the intention to have the users require to change this every time

pedrolamas, 2023-12-11:
Hi @morriscode, thank you for opening this ticket.

The only reason Fluidd would try using port `4408` is because your browser opened Fluidd on port `4408` (which is perfectly fine).

Fluidd is a Progressive Web App (PWA) served by nginx, so when you open Fluidd in your browser, you are really accessing nginx, and it in turn serves Fluidd files to the browser.

Your nginx configuration seems to also serve then "/webcam" content on the same port, but the fact you are experiencing this issue tells me that there is some sort of nginx misconfiguration.

The "/webcam" section in nginx configuration should look something like this:

https://github.com/pedrolamas/docker-iot-stack/blob/59467c8fb9d14855fea1f8f413ccdcccd9e2999d/nginx/etc-nginx-templates/nginx.conf.template#L282-L289

morriscode, 2023-12-12:
Interesting take. 

This was installed via the helper script [here](https://github.com/Guilouz/Creality-K1-and-K1-Max). which installs moonraker and nginx i believe. 

the instructions say to only use 4409 if you use mainsail, which i am not. 
All of this was working fine until I updated Fluidd, which is the same issue other on reddit have posted.  Are you saying it's more likely that we all simultaneously changed our nginx configs, or they were changed as part of a separate upgrade?

pedrolamas, 2023-12-12:
I am saying that the script that generated the nginx bindings for the webcams is incorrect.

I don't know the specific script, but I took a quick look at it and could see that all it does is a redirect with defaults applied, which is insufficient when we want to stream mjpeg data.

pedrolamas, 2023-12-21:
As this seems to be related to a 3rd party, I am now closing this.

closed 2023-12-21 by pedrolamas