fluidd-core/fluidd #1282: Process uptime

thijstriemstra, 2023-12-19:
### Is your feature request related to a problem? Please describe

I'd like to know and see the uptime for:

- fluidd
- klipper
- moonraker
- host running fluidd
- any of the mcu's

### Describe the solution you'd like

Add the info to the `/system` section.

### Additional information

fluidd 1.27

pedrolamas, 2023-12-19:
Hi @thijstriemstra thanks for opening this Feature Request.

Before we can do anything on Fluidd side, we will need Moonraker to provide that data, which I believe at the moment it does not.

thijstriemstra, 2024-02-20:
i opened https://github.com/Arksine/moonraker/issues/817