fluidd-core/fluidd #1286: Tool controls missing tooltips

thijstriemstra, 2023-12-21:
### Fluidd Version

1.27

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

Most of the buttons on the tool controls do not have tooltips.

[image]

### What did you expect to happen

Have consistent tooltips appear, not only for 2 or 3 buttons.

### How to reproduce

Roll over buttons.

pedrolamas, 2023-12-21:
Hi @thijstriemstra, thanks for opening this issue.

We have added tooltips to the two buttons that share the same icon but in order for users to differentiate what each on does (so we have a "Home X and Y" and "Home Z")

I will look into adding more tooltips (though it does seem a bit unnecessary), but ideally, I would prefer for us to have icons that actually show what is being "homed", and I will look into that possibility.

thijstriemstra, 2023-12-21:
> I would prefer for us to have icons that actually show what is being "homed",

Exactly, I often mistakenly press the "UP" button for Z when I want to home Y. A tooltip would 100% prevent me from doing this.

pedrolamas, 2023-12-21:
Unrelated (but as I am not sure if you've noticed), we now have 2 more options for toolhead controls: bars and circle (change in the settings)

thijstriemstra, 2023-12-21:
> I will look into adding more tooltips (though it does seem a bit unnecessary),

If you think they're not needed, perhaps they should be removed from all the controls. Now only 30% of controls have a tooltip, making it look inconsistent (e.g. buggy). I think it's either all with or all without.