fluidd-core/fluidd #1290: Simplify finding MCU IDs

iamthesoy, 2023-12-24:
### Is your feature request related to a problem? Please describe

With preconfigured OS images and pre-flashed boards becoming more and more common in printer kits, the only reason to SSH into the host is to run `ls /dev/serial/by-id/* ` It works fine, but it would be cool if this wasn't needed, and would simplify the process for newbies.

### Describe the solution you'd like

I was thinking that when editing .cfg files, an option in the right click menu that lists possible connected MCUs (which can be inserted when clicked on) would be a reasonably elegant solution - see poorly made mockup image
This could be set to only appear on lines starting with `serial: ` to reduce clutter.
[image]

pedrolamas, 2024-01-03:
Hi @iamthesoy, thank you for submitting this feature request.

Fluidd is just a web app that runs in the browser (client side) and uses Moonraker as the server-side data provider.

In order for Fluidd to implement such feature, Moonraker will first have to provide this data to Fluidd.

pedrolamas, 2024-02-01:
> FWIW this just got merged: [Arksine/moonraker#798](https://github.com/Arksine/moonraker/pull/798)

Assuming these changes indeed provide the data we require, I guess we can now indeed start working on providing this feature! 🙂

pedrolamas, 2024-02-01:
I just checked... and I'm impressed!!

[image]

PR #1383 (2024-03-13): feat: adds support for machine peripherals
Adds support for the new Moonraker Machine Peripherals (Devices)

CAN bus tab will appear if one or more CAN interfaces exist -> this bit has not been thoroughly tested as I do not have a CAN network myself that I can use for testing purposes.

## Shortcut on system info card

[image]

## Shortcut on file editor

[image]

## On open

[image]

## After refresh, no devices found

[image]

## After refresh, serial devices found

[image]

## After refresh, usb devices found

[image]

Resolves #1290