fluidd-core/fluidd #1292: Smarter Calibration Function to avoid Bed Damage

Hammerfest, 2023-12-25:
### Is your feature request related to a problem? Please describe

New 3D Printer user so mistakes where made.
I adjusted my bed as it was not level by a large margin, all I did was reset screw positions but that changed the bed by 3mm~
I hit clear for the bed mesh then calibrate in Fluidd.
You can imagine what happened with an unexpected 3mm rise in the bed, the hot end started role playing as a CNC on the plate and other unsavory sounds/damage.

### Describe the solution you'd like

The auto leveling function on some 3D printers does homing first, then does "leveling".
While I am not requesting the FULL leveling procedure every time, it would be a good idea to lower the bed 5-10mm, move the print head to the center, then check for home discrepancy, if found, proceed to home THEN calibrate.

pedrolamas, 2023-12-25:
Hi @Hammerfest, thank you for submitting this Feature Request.

I should however make you aware that Fluidd is just a web interface for Klipper (the firmware) communicating via Moonraker (the server side API), so we have no control over how homing is performed.

Having said that, you can override the homing order in Klipper (please check Klipper docs for that).

Hammerfest, 2023-12-25:
Correct, seeing as the Creality software is also just a front end for klipper yet it manages to do the same they must be doing something that can be done on the front end much like in Fluidd

pedrolamas, 2023-12-25:
I have no idea what Creality does (they run unsupported forks of Klipper, Moonraker, Fluidd and others) but I would expect them to have changed the Klipper configuration to conform to what you described.

Nevertheless, this is not a Fluidd feature to implement as we have no control whatsoever from our side to change the homing sequence.

Hammerfest, 2023-12-25:
I think your hyper focusing on the changing the homing sequence I mentioned at first, there is still SOMETHING fluidd could do to at least warn of this kind of situation, it is a front end UI.

Hammerfest, 2023-12-25:
Actually, I will close this request and open a new one with no mention of changing the klipper sequence just changing the way the front end behaves based on specific markers which is within a front ends capability's.

closed 2023-12-25 by Hammerfest