fluidd-core/fluidd #1294: Fluiddpi asking to add printer when only one printer

samisanerd1, 2023-12-26:
### Fluidd Version

1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

i have had my ender 3 v2 and raspi 4 setup for a month now. it was working great until, right after i install the sprite pro direct drive extruder, the thing does this. i have no idea what to do i'm not sure how to change my config files, because it only shows [image]

### What did you expect to happen

The ui to open and for me to access my printer

### How to reproduce

Im not sure. it just happened

pedrolamas, 2023-12-26:
Hi @samisanerd1, from your message on Klipper Discord I see that you have found and fixed this problem, so I am closing this now.

closed 2023-12-26 by pedrolamas

samisanerd1, 2023-12-26:
I didnt message any discord, but i did manage to fix it by reinstalling everything.

pedrolamas, 2023-12-26:
My apologies @samisanerd1 , somehow I saw a very similar screenshot in Discord today and for a moment thought it was from the same user.