fluidd-core/fluidd #1295: Repo is corrupt

thijstriemstra, 2023-12-27:
### Fluidd Version

1.27.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I'm seeing this in the update screen:

[image]

the `Repo is corrupt` message starting showing up today.. These are pristine clones of the official klipper/moonraker repos.

### What did you expect to happen

Not see a warning message.

### How to reproduce

-

pedrolamas, 2023-12-27:
Hi @thijstriemstra, those details are provided by Moonraker, all Fluidd does is present the status as it gets it.

Your moonraker.log should have some information about what happened (probably some git error messages there)

thijstriemstra, 2023-12-27:
I rebooted everything and checked  moonraker logs:

```
tail -f -n 100 printer_data/logs/moonraker.log
```

But no errors and UI also seems to be fixed now.. weird? I'll close it for now and re-open when I see it again.

thijstriemstra, 2023-12-27:
What are the conditions for that `Repo is corrupt` warning to show up btw?

closed 2023-12-27 by thijstriemstra

pedrolamas, 2023-12-27:
Only condition from Fluidd perspective is for Moonraker to report the repos as corrupt.

I'm not entirely sure what conditions does Moonraker have to consider the repo corrupt.