fluidd-core/fluidd #1297: Thermal device toggle reset after klipper restart

thijstriemstra, 2023-12-29:
### Fluidd Version

1.27.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

For some reason whenever Klipper is restarted, a toggled thermal device is untoggled. Highly annoying because I have to toggle it after every reset (since I don't want it to show up in the graph).

### What did you expect to happen

Remember toggled items.

### How to reproduce

Toggle thermal device, restart klipper and it'll be untoggled again.

Dycus, 2024-06-10:
I also experience this and would really love if the thermal device view state was always persistent. I have 13 temperature sensors but only want to graph 5 of them, so it's a bit of clicking to reset the graph every time I restart the firmware or turn the printer on. :) (extremely mild inconvenience)

Once in a while I can get a random disconnection from Moonraker, usually after coming back to the printer after an overnight print. This also results in the thermal view state resetting without a firmware restart.

PhilippMolitor, 2024-07-17:
Same for me. It only happens when I reload fluidd via the UI. Once i refresh the browser window, the correct settings are re-applied. Seems to be related to the global store retrieving the settings data.

thijstriemstra, 2024-10-20:
Is this a moonraker-limitation thing or unclear how to implement this @pedrolamas? not sure what such a FR entails here.

pedrolamas, 2024-10-20:
The thing is Moonraker does not keep aggregated per-file statistics!

Currently, we get the history entry (print orders indicating which file was printed, when, and a bunch of metadata), but we don't ask ALL the entries from Moonraker to keep memory usage under control.

So we can calculate per-file stats from that subset of history data we get from Moonraker, but ideally it would be great if Moonraker can provide that (which currently I don't think it can).

thijstriemstra, 2024-10-21:
reproducing it shouldn't be hard at all, i'll doublecheck how to later today.

PhilippMolitor, 2024-10-22:
@thijstriemstra I think for me it happened almost always when restarting klipper. i use a moonraker power device to enable my 24V PSU, and I toggle that with the UI switch. It also controls the klipper service when set up correctly in moonraker.

thijstriemstra, 2024-10-22:
> reproducing it shouldn't be hard at all, 

Yep, untoggling heaters and restarting Klipper (after a config change) is enough to reproduce this bug.

pedrolamas, 2024-10-24:
Managed to reproduce this consistently, will work for a fix for the next release!

closed 2024-10-25 by pedrolamas

PhilippMolitor, 2024-11-08:
for me this bug still happens from time to time, not sure under which conditions..

pedrolamas, 2024-11-08:
I am currently doing some work on that area but have yet to see that fail!

I will keep my eyes on top of this and run a few more tests, but without being able to consistently reproduce, it will be very tricky to change anything...

PhilippMolitor, 2024-11-08:
i edited my config a lot and had a lot of restarts of the firmware and the klipper service, maybe it happens when the connection just breaks away because of that..

pedrolamas, 2024-11-08:
> Ok, I just managed to reproduce it by restarting Moonraker... which I had not tested before, only with Klipper!

I have just pushed a fix for the above scenario: https://github.com/fluidd-core/fluidd/commit/24d0db9515f20fc7715c3b01fecd70ddf88489e4

PR #1514 (2024-10-25): fix(Charts): allow resetting chart store only
Ensures we ONLY clear the chart data on reset, and not the whole chart state (that includes the selected sensors and heaters!)

Fixes #1297