fluidd-core/fluidd #1304: Live speed is missing from Fluidd

RenanAz, 2024-01-10:
### Fluidd Version

1.27.1

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

The Live speed feature (added here [)](https://github.com/fluidd-core/fluidd/pull/1019) is now missing from below the live coordinates. 
[image]

### What did you expect to happen

Expect to see the live printing speed as it was in the past releases.

### How to reproduce

Start a print and observe below the live coordinates.

RenanAz, 2024-01-10:
Ah ok, it makes sense. Thank you!
Just out of curiosity, was this in the release notes and I missed it? I did try to search for it before. 
Thanks again for your awesome support.

pedrolamas, 2024-01-11:
It was the "improves estimates calculation" (granted, not very explicit to what we actually did there!) https://github.com/fluidd-core/fluidd/pull/1283

closed 2024-01-11 by pedrolamas