fluidd-core/fluidd #1309: Add exclude object map

sendorm, 2024-01-12:
### Is your feature request related to a problem? Please describe

Hello, mainsail has a feature where you can select the object to be excluded from a map of objects in the print.
With the current method of just creating a list with object it is very hard to exclude the object you want as for instance when you split a print into object into parts with orcaslicer, the slicer just creates new object with incremented ids. 

### Describe the solution you'd like

This is just a Quality of Life Update. The solution would be adding a "map" to show the object in the current print, so that the item to be excluded can be chosen easier. 

### Describe alternatives you've considered

Besides using the mainsail ui, I couldn't find a solution. Also I can split the object into sub object from orcaslicer to find the id needed.

pedrolamas, 2024-01-12:
Hi @sendorm, you can exclude any object from the Gcode Preview card, it's a feature we've had since v1.19.0, added via #754

closed 2024-01-12 by pedrolamas

sendorm, 2024-01-12:
Thanks for the information.