fluidd-core/fluidd #1313: Link the update information to the documentation

Sonnenwerk, 2024-01-18:
### Is your feature request related to a problem? Please describe

I would like to have an easy clickable link to the homepage of a klipper addon, so that i don´t have to search so much.
addon: Preferable i would like to have a direct link to the "what to do when you upgrade" notes.
Currently i manually add useful pages/links to the apropriate .cfg files as a remark. 

### Describe the solution you'd like

[image]

### Describe alternatives you've considered

I have some variants
a) only the link to the mainpage of the project is displayed (the information is already in the moonraker.conf)
b) It is differentiated between "main project page" and the "do i have to do something specific for this update" link
c) we don´t do it on the softwareupdate page. instead we add a new tab under "configuration "other data" and add the feature to manually add some more wanted perma links too, like a link to the hardware vendor or some good calibration guide...

pedrolamas, 2024-01-23:
Hi @Sonnenwerk, thank you for sending this Feature Request.

The way that the Moonraker update manager works is by pulling changes from the git remote that might not (and probably aren't) associated with a tagged release, so there is no concept of "release notes" in this process.

The only thing close to it is the list of commits (changes) that have happened since the version you have and the remote version, which we already show - in fact, we add links to each of these commits to show the changes directly on the git repo.

To have some sort of "summary" added, we will need for Moonraker support as I don't think we should do assumptions just from Fluidd side.

Sonnenwerk, 2024-01-23:
Yes, summarized release notes and update orders are in the far future (i guess).
I am aware of that. It was just an outlook, what i have in my mind, that could come later and would be one of my wishes.

The one - a big step easier - and not so high sophisticated is the option a) which already would help me somewhat.

I post a part of my printer.cfg to show, what i currently do in there. My Voron commentary page is already one page long, consisting of tipps and tricks and links to all the hardware pages, i need for configuration and install.
------------
#CB1 eMMC 1.1
  #main git page  https://github.com/bigtreetech/Manta-E3EZ/tree/master
  #user manual    https://github.com/bigtreetech/Manta E3EZ/blob/master/BIGTREETECH%20Manta%20E3EZ%20V1.0%20User%20Manual.pdf
#TFT35 2.1 SPI
#update mcu:
  #Show serial device ls /dev/serial/by-id/*
  #make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_5A003A000C504B5735313920-if00
[include macro_ender.cfg]
[include fluidd.cfg]
#[include neopixe
-----------
Not sure, if i found the right words. I wanted to say, that i have "some" need adding links to other pages. And sounded some different ideas in this FR.

PR #1348 (2024-02-09): feat: adds link to components in software updates
Adds a link to components in the Software Updates panel. When clicked, it will open the repository URL on a new separate browser window/tab.

[image]

On hover, a tooltip with the URL is shown.

[image]

Resolves #1313
closed 2024-02-09 by pedrolamas