fluidd-core/fluidd #1315: Camera support proper Moonraker API -- like Mainsail does

Ltek, 2024-01-20:
### Is your feature request related to a problem? Please describe

Cameras added in Fluidd do not work in any other user interface.  Unlike cams added in Mainsail or Moonraker (which both use the Moonraker API) these added/deleted in Mainsail or Moonraker are available in ALL user interfaces.

### Describe the solution you'd like

Fluidd to use the Moonraker API

### Describe alternatives you've considered

Non exist - must create cameras in Mainsail.

pedrolamas, 2024-01-20:
Hi @Ltek , can I ask what version of Fluidd did you experience this on, and that device are you using has Klipper host (raspberry pi or other)?

Ltek, 2024-01-25:
> Hi @Ltek , can I ask what version of Fluidd did you experience this on, and that device are you using has Klipper host (raspberry pi or other)?

 always update to the latest when the UI notifies me, I dont keep track. Runs on Creality K1

pedrolamas, 2024-01-25:
Thank you @Ltek can you please take a screenshot of the Updates section in the Fluids Settings and post it here please?

Also, is your K1 currently rooted?

Ltek, 2024-01-25:
> Thank you @Ltek can you please take a screenshot of the Updates section in the Fluidd Settings and post it here please?
> 
> Also, is your K1 currently rooted?

I'm not sure what you are looking for... based on my testing - working with the developer of MobileRaker... he stated, and we seemed to confirm, that Fluidd -does not- have support for the Moonraker API. If it does, its broken and has been for many versions of Fluidd. You should easily be able to repro ...

Create cam in Mainsail -> shows in Fluidd
Create cam in Fluidd -> does not show in Mainsail or Moonraker

my K1 is rooted or I'd never be able to use Fluidd or Mainsail with it.

PR #1386 (2024-03-05): refactor: Moonraker Webcam API
Replaces the legacy direct database access to the "webcams" namespace with the Moonraker Webcam API.

On first launch, we will still do the migration step of taking the old "cameras" data in the "fluidd" namespace and sent it to the "webcams" namespace, but that will then require a manual restart of Moonraker (which I have not included here)

Includes a new check for minimum Moonraker and Klipper versions, that will display a notification if the current version does not satisfy the required minimum.

[image]

[image]

Resolves #1315
closed 2024-03-05 by pedrolamas