fluidd-core/fluidd #1317: Unable to download logs when using auth/authz with moonraker

samwiseg0, 2024-01-21:
### Fluidd Version

fluidd: v1.27.1-2056453

### Browser

Chrome, Safari

### Device

Desktop PC

### Operating System

macOS

### What happened

When trying to download log files from fluidd while using authz with moonraker. Files cannot be downloaded. This has been an issue for some time but I have not had time to file this bug. 

In safari, a dummy file is downloaded with error:
```{"error": {"code": 401, "message": "Unauthorized", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1767, in _execute\n    result = self.prepare()\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 578, in prepare\n    self.current_user = auth.check_authorized(self.request, self.endpoint)\n\n  File \"/home/pi/moonraker/moonraker/components/authorization.py\", line 807, in check_authorized\n    raise HTTPError(401, \"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)\n"}}```

In chrome an error will popup. 

After adding a client as trusted there is no issue downloading files. This could be a moonraker issue but I am unsure as the request is coming from fluidd?

### What did you expect to happen

The log file to be able to be downloaded while having authz configured in moonraker. 

### How to reproduce

Configure moonraker to prompt non trusted clients to auth. 

Navigate to any log file in the printer logs dir. Try and download.

matmen, 2024-01-21:
I can't reproduce this in either Firefox or Chrome on Windows. Can you share your moonraker.log, fluidd-access.log and your moonraker.conf (either by turning forced logins off or using scp)?

samwiseg0, 2024-01-21:
Sure here is all the logs. 
[unable_to_download_files.zip](https://github.com/fluidd-core/fluidd/files/14002687/unable_to_download_files.zip)

samwiseg0, 2024-01-21:
Just for clarity. I do not have `force_logins` set to true. I just omitted my client from the trusted clients list.

matmen, 2024-01-21:
Ahh, thanks for the clarification! I can reproduce it now, I'll take a look at it

samwiseg0, 2024-01-21:
> Ahh, thanks for the clarification! I can reproduce it now, I'll take a look at it

Awesome! Thank you!

closed 2024-01-22 by matmen