fluidd-core/fluidd #1323: incorrect spoolman url

thijstriemstra, 2024-01-25:
### Fluidd Version

1.27.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

When I click 'Manage spools' it takes me to `http://0.0.0.0:7912/` which cannot be resolved. Spoolman runs on same host as fluidd.

### What did you expect to happen

It should take me to a custom URL or the same domain fluidd runs on (e.g. `http://myprinter.local:7912/`)

### How to reproduce

Use spoolman manage spools button.

matmen, 2024-01-26:
Thanks for your report. A simple fix for this would be to set an externally reachable address as the server address in your moonraker config:
```
[spoolman]
server: http://myprinter.local:7912
```

Unless I am missing something, I don't really see a need to have a separate input box in the UI for this - setting the backend URL in a single place (i.e. the moonraker config) seems more reasonable.

thijstriemstra, 2024-01-27:
Totally missed that option, embarrassing ;)

closed 2024-01-27 by thijstriemstra