fluidd-core/fluidd #1337: Ability to choose option at print startup

oRudi07, 2024-01-31:
### Is your feature request related to a problem? Please describe

Is there a way to program fluidd to query before starting a print? In Start_macro I have the options to run a completely new Bed_Mesh or load a specific profile or just create a Bed_mesh within the boundaries of the part to be printed. Currently I'm commenting out the options that are not wanted at startup. It would be nicer if a window would open when I started printing where I could select the option

### Describe the solution you'd like

nothing

pedrolamas, 2024-01-31:
Hi @oRudi07 , macro prompt dialogs are supported in Fluidd, though this requires that you have the latest Fluidd and fluidd-config (or mainsail-config).

Please check #1267 for examples and details.

oRudi07, 2024-01-31:
Hi, thats what i am looking for.
But I need some help to get this in my config. I wrote these macros:
[gcode_macro SHOW_OPTION_BED_MESH]
gcode:
    RESPOND TYPE=command MSG="action:prompt_begin BedMesh"
    RESPOND TYPE=command MSG="action:prompt_text How to level?"
    RESPOND TYPE=command MSG="action:prompt_button_group_start"
    RESPOND TYPE=command MSG="action:prompt_button Load Default|LOAD_SAVED_BM|info"
    RESPOND TYPE=command MSG="action:prompt_button whole bed|BM_WHOLE_WHOLE|info"
    RESPOND TYPE=command MSG="action:prompt_button only part|BM_ONLY_PART|info"
    RESPOND TYPE=command MSG="action:prompt_button_group_end"
    RESPOND TYPE=command MSG="action:prompt_button_group_start"
    RESPOND TYPE=command MSG="action:prompt_button CANCEL|CANCEL_PRINT|warning"
    RESPOND TYPE=command MSG="action:prompt_button_group_end"
    RESPOND TYPE=command MSG="action:prompt_show"

[gcode_macro LOAD_SAVED_BM]
    BED_MESH_CLEAR
    BED_MESH_PROFILE LOAD="default"
    

[gcode_macro BM_WHOLE_WHOLE]
     BED_MESH_CLEAR
     BED_MESH_CALIBRATE
     BED_MESH_PROFILE SAVE=x1
    

[gcode_macro BM_ONLY_PART]
    BED_MESH_CLEAR
    BED_MESH_CALIBRATE PRINT_MIN={params.PRINT_MIN} PRINT_MAX={params.PRINT_MAX} FORCE_NEW_MESH=True
    BED_MESH_PROFILE SAVE=x1

Then, in START_PRINT, I call SHOW_OPTION_BED_MESH. The window appears with configured buttons, but after ckicking one option nothing happens -and during the window appears, theSTART_PRINT macro continues to be processed

What else do I have to do?

pedrolamas, 2024-01-31:
I would say that you now need 2 changes:

- on the `PRINT_START` macro, add a `PAUSE` as the last command (so after  `SHOW_OPTION_BED_MESH`)
- on each of the option macros, add a `RESUME` to continue the print.

Note: I have not tested this, but in theory, it should work...

oRudi07, 2024-01-31:
Thanks for the answer. One more question: how to get the window closed after clicking an option?

pedrolamas, 2024-01-31:
Use `RESPOND TYPE=command MSG="action:prompt_end"` to close the dialog.

oRudi07, 2024-01-31:
also found out in the meantime.
Thank you very much

pedrolamas, 2024-02-01:
Given the above, I think we got a good compromise for this feature by using the macro prompt dialogs!

As such, I am now closing this ticket.

closed 2024-02-01 by pedrolamas

oRudi07, 2024-02-01:
One more thing: If i want to pause a makro (pause the gcode from processing), i cannot use PAUSE because thie is the macro pause....

pedrolamas, 2024-02-01:
There is a limit to what is possible with the current Klipper provided support...

Anything more than the described above, I think we will need upstream support added directly in Klipper.