fluidd-core/fluidd #1349: Change Configuration Reference Page Based on Klipper forks

Kizime123, 2024-02-09:
### Is your feature request related to a problem? Please describe

I currently use dangerklipper for all my klipper needs,  it's a bit more of an inconvenience to find configuration reference details since I often will click the configuration reference button. This leads me to realize I have to actually go into the dangerklipper github for the proper configuration reference.

### Describe the solution you'd like

Whenever someone is using a klipper fork instead of referencing the stock klipper configuration reference page (https://www.klipper3d.org/Config_Reference.html), reference the forks configuration reference page.

pedrolamas, 2024-02-11:
Hi @Kizime123, thank you for submitting this Feature Request.

I have taken a quick look at this, and my first thought is that we can't perform any "automatic" discovery just by looking at the fork the user has (there is no guarantee that the docs from said fork have been published!), so manual (user) indication is indeed required here, probably via an optional setting in Fluidd.

Kizime123, 2024-02-12:
@pedrolamas This change would make alot of sense. Knowing docs might not be updated instead a setting to allow you to manually enter it would make sense! Could it be possible to potentially have the fork actually set this information too on some sort of input too?

PR #1446 (2024-06-03): feat: adds Danger Klipper config support
Checks if we are running on [Danger-Klipper](https://dangerklipper.io) and if so, use its website for documentation.

Note: given the popularity of Klipper, there are now quite a few "flavors" proliferating, but we have **no plans to add indiscriminate support to any of these Klipper forks.**

However, Danger Klipper is now the main Klipper fork and is gaining significant traction, hence why we are specifically adding config support for it here.

Resolves #1349
closed 2024-06-03 by pedrolamas