fluidd-core/fluidd #1350: FR: Seperate options for on/off switch in top bar

Arakon, 2024-02-09:
### Is your feature request related to a problem? Please describe

The on/off switch only allows to power i.e. the printer on or off. If one was to use a power down macro (in this specific case, to switch neopixel LEDs off before shutting off the relay the cuts power to the mainboard and power supply, while leaving the Raspberry running on its own PSU), then the user has to click on the macro's button instead of using the power button in the top bar to shut down properly. Considering how many things Klipper can control these days, it would be nice to have the option to perform extra steps before power is cut hard.

Unfortunately, assigning the button to a klipper_device will not allow the user to also use the button to turn the printer back on anymore.

### Describe the solution you'd like

Have seperate on/off entries of the power button. Example:
- Power is off, click the button to toggle the relay GPIO pin via Moonraker
- Power is on, click the button to execute a macro of choice, i.e. to disable the lights on a display or tell the display to shut down, send a message, park the printhead, and/or toggle GPIO pins.

pedrolamas, 2024-02-11:
Hi @Arakon, thank you for submitting this Feature Request.

I haven't delved much into it yet, but have you considered using a Moonraker `[power]` device with `type: klipper_device` and calling a macro?

Please check [here](https://moonraker.readthedocs.io/en/latest/configuration/?h=button#klipper-device-configuration) for more information on this.

Arakon, 2024-02-11:
I have. Unfortunately, when using klipper_device as a type, I can not turn the printer back on, since Klipper goes into shutdown state once the mainboard connection is lost, so no further communication with Klipper is possible (i.e. to run a macro to turn on power).