fluidd-core/fluidd #1351: Spoolman title bar incorrectly rendering

thijstriemstra, 2024-02-09:
### Fluidd Version

1.28

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

Clicking "Change spool" shows a broken header:

[image]

### What did you expect to happen

Normal panel title

### How to reproduce

Use spoolman 0.16.1 and fluidd 1.28.0

Cyondil, 2024-02-10:
Also seeing the same here.

[image]

pedrolamas, 2024-02-10:
Hi @thijstriemstra and @Cyondil, thank you for reporting this issue.

My apologies for this, I missed it on a refactor I did, but this will be fixed on the next release v1.28.1

PR #1354 (2024-02-10): fix(spoolman): dialog title
Adds missing Vue.js expression brackets.

[image]

Fixes #1351
closed 2024-02-10 by pedrolamas