fluidd-core/fluidd #1359: Exclude object not working on mobile

viesturz, 2024-02-15:
### Fluidd Version

1.27.1

### Browser

Chrome

### Device

Mobile device (Phone, Tablet or similar)

### Operating System

Android

### What happened

Nothing happens when I tap the exclude icon

### What did you expect to happen

The confirmation popup should show up

### How to reproduce

Enable exclude object in Klipper, print multi part gcode.
Open the guide preview and try to exclude one object.

### Additional information

Works fine from desktop on the same setup.

thijstriemstra, 2024-02-15:
> 1.27.1

I'd give the latest (1.28) a try as well before confirming this bug report.

pedrolamas, 2024-02-15:
Hi @viesturz, thanks for reporting this.

Can I ask that you try again on the phone in an anonymous browser tab?

If it works on the anonymous browser tab, please clear the browser cache to fix the problem.

matmen, 2024-02-15:
I can reproduce it on Chrome on Android, looks like a "touch canceled" event is being fired immediately after the "touch start" event is triggered. Honestly looks like a Chrome bug to me, but I'll investigate tomorrow.

pedrolamas, 2024-02-16:
FWIW, I also managed to reproduce this issue with Edge on Android.

PR #1366 (2024-02-18): refactor: uses vue2-touch-events for tap
To fix the tap issues, we are now introducing the vue2-touch-events library.

This will deprecate #1363

Fixes #1359
closed 2024-02-18 by pedrolamas