fluidd-core/fluidd #1362: List of feature requests, missing features, some bugs

yw4z, 2024-02-16:
### Is your feature request related to a problem? Please describe

Sorry for long list but i didn't want to spam issues section. I want to share my experience and ideas for Fluidd. Some might be opposite direction for your road map so i understand if you want to skip most of them. no problem just sharing

Quality of life Improvements
**Dashboard > General**
• Clicking empty area on title bar to expand / roll, just arrow button expands / rolls currently
**Dashboard > Fans & Outputs >**
• Distribute objects to columns by height instead of number. using number as calculation gives not equal spacing for columns. you can see there is dead space on left
[image]

• Hiding or showing elements from fluidd UI would be useful without editing config file. i guess this one and next one is creality sepesific because something will broke if i change something in printer config
• Rename elements from fluidd UI would be useful without editing config file.
	
• Macro support on here can be quite useful. i tried to add 3 switches for toggling adaptive purge, adaptive mesh and calibration in printer config but i have to give them pins in config and there is not much empty pins on my board.
>How it will works
>>Create a custom switch from Fluidd UI
>>Switch will change gcode variable
>>Will check regularly gcode variable or config object to get status
i can do this with macro but macro buttons doesnt show any info about status or value of variable
	
**Dashboard > Thermals**
• Again ability to hide / show elements
• Alert when temp drop / reach to target. it would be useful when printing ABS or drying filaments. also might be useful if Fluidd support browser notifications for that

**Dashboard > Tools**
• Heat extruder button to enable retract / extrude
• Toolhead controls has dead space on right side. if i reduce width of window a bit it moves controls to next row
[image]
[image]

**Dashboard > Macros**
• Ability to re-order categories. Currently i use numbers at beginning for easy hack
• Macro organizing > Dropdown like buttons to grouping similar macros.
	i recomend dropdown method as alternative because popups appears center of screen and increases mouse movements for users
>example usages
>>Button Name: Material , Dropdown buttons: Load, Release
>>Button Name: PID Tuning , Dropdown buttons: Bed, Hotend
>>Button Name: Calibration , Dropdown buttons: Input Shaper, Belts, Steppers
>>Button Name: Silent Mode , Dropdown buttons: On, Off
>>Button Name: Set Chamber Teperature , Dropdown buttons: 45(PLA 4H), 65 (ABS PETG >3H), 55 (TPU/TPE >4H)
>>Button Name: Input Shaper , Dropdown buttons: On, Off

This type of organiation can save a lot space for macros

**Console Page**
• Auto focus on input box after clicking console button on left sidebar
• Ability to run multi line jinja code from input box. Also same feature for dashboard console

**History Page**
• Printer Stats > total filament used in gr/kg

**Tuning Page**
• Configurable input shaper values
• Macros section. it could be nice to see calibration macros in here. maybe would be an option on category settings for "Add  Tuning Page"

**File Manager**
• Create zips with Date Stamp 2024-02-12-1315-41315 (best for keeping backups and archiving)
• Download option for multiple selections. i saw some webdaw based local file sharing apps can do this

**Text editor**
• Right click or a play button next to gcode_macro name to run command without going console. this one can really helpful while developing gcodes
• Run item on right click menu for selected lines for gcode_macro

**Might be bug**
• Fluidd not checking fan speeds after firmware restart
• Updating fluid itself deleting custom logos. i installed creality logo with guilouz script.  you can find logos in here
https://github.com/Guilouz/Creality-K1-and-K1-Max/tree/main/Scripts/files/fluidd
i guess requesting adding creality to logos  more sensible request instead of keeping custom logos folder :)

Also i want to add i want to contribute the project for UI and UX design and experience, i can make templates or icon designs when it needed if you interested

### Describe the solution you'd like

*

pedrolamas, 2024-02-19:
Some more comments on the topic:

> **History Page** Printer Stats > total filament used in gr/kg

Fluidd is just a web frontend, we rely on Moonraker to provide these kind of statistics, and at the moment, this is not supported on the [server/history/totals](https://moonraker.readthedocs.io/en/latest/web_api/#get-job-totals) endpoint.

> **Might be bug** Fluidd not checking fan speeds after firmware restart

I'm not sure I understand this one... can you please elaborate on this?

> **Might be bug** Updating fluid itself deleting custom logos. i installed creality logo with guilouz script. you can find logos in here [Guilouz/Creality-K1-and-K1-Max@`main`/Scripts/files/fluidd](https://github.com/Guilouz/Creality-K1-and-K1-Max/tree/main/Scripts/files/fluidd?rgh-link-date=2024-02-16T20%3A43%3A53Z)

This method is an unsupported hack, so not a bug. The only supported way for custom theming is via [Custom Styling](https://docs.fluidd.xyz/customize/themes#custom-styling).

> i guess requesting adding creality to logos more sensible request instead of keeping custom logos folder :)

Creality has proven multiple times that they have no problems with taking open-source code (eg. Klipper, Moonraker, Fluidd, Mainsail, KlipperScreen), changing it, and then using it on their printers without releasing said changes (thus breaking the open-source licenses). We've seen this with the Sonic Pad, and we are still seeing this today with other Creality products.

So until the day Creality changes the way they approach open-source, this is not going to happen.

> Also i want to add i want to contribute the project for UI and UX design and experience, i can make templates or icon designs when it needed if you interested

This is an open-source project and user contributions are very welcome!

yw4z, 2024-02-19:
Hi @pedrolamas thanks for sharing details now i understand limitations

Thanks for adding Timestamp and autofocus on console page features

> Fluidd not checking fan speeds after firmware restart

Fan speeds shows 0 value on Fluidd UI after firmware restart but chamber fan spinning at %100. i'm not sure how you checking values in intervals or klipper reports to fluildd. i can share klipper log if necessary

> File Manager Download option for multiple selections. i saw some webdaw based local file sharing apps can do this

i can suggest you can automatically zip and download on multiple selections as a workaround. probably it will be easier to deal with multiple file download

i got your point on Creality

Should i use discussions or issue section for sharing design templates. i made a template for dashboard > tools section. mostly reorganized things and improved readability

yw4z, 2024-02-19:
also i updated 28.1 and tried autofocus on console but its not working if "flip layout" option selected. works as expected on normal layout

pedrolamas, 2024-02-19:
> also i updated 28.1 and tried autofocus on console but its not working if "flip layout" option selected. works as expected on normal layout

Probably because I forgot to add for both situations! 🙂

We'll be sure to fix it for the next release.

yw4z, 2024-02-19:
No need to hurry just reporting :)

pedrolamas, 2024-03-17:
@yw4z the "workable" items on this list have made it to the latest 1.29 release, so I will be closing this ticket now.

From our perspective, it is not feasible to prioritize and track work items when they are on a single ticket, but nevertheless I understand this was a good opportunity for you to put all your suggestions in writing (something I really appreciate) and I believe I have replied to most of them above.

closed 2024-03-17 by pedrolamas

yw4z, 2024-03-17:
@pedrolamas thanks for all your interest