fluidd-core/fluidd #1385: Incorrect System Information (Distribution) for Armbian

halfmanbear, 2024-03-05:
### Fluidd Version

v1.28.1

### Browser

Chrome

### Device

Laptop

### Operating System

macOS

### What happened

Distribution value incorrect from DistroInfo,
[image]

### What did you expect to happen

[image]

### How to reproduce

armbian-release files always start with `# PLEASE DO NOT EDIT THIS FILE`

```
mks@znp-k1:~$ cat /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=znp-k1
BOARD_NAME="ZNP-K1"
BOARDFAMILY=rockchip64
BUILD_REPOSITORY_URL=https://github.com/halfmanbear/Armbian-ZNP-K1-build.git
BUILD_REPOSITORY_COMMIT=9e04b40
LINUXFAMILY=rockchip64
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
FORCE_BOOTSCRIPT_UPDATE=
VENDOR="Armbian-unofficial"
VENDORDOCS="https://docs.armbian.com/"
VENDORURL="https://duckduckgo.com/"
VENDORSUPPORT="https://community.armbian.com/"
VENDORBUGS="https://armbian.atlassian.net/"
BOOTSCRIPT_FORCE_UPDATE="no"
BOOTSCRIPT_DST="boot.cmd"
VERSION=24.2.0-trunk
REVISION=24.2.0-trunk
BRANCH=current
```
```
mks@znp-k1:~$ uname -a
Linux znp-k1 6.1.67-current-rockchip64 #2 SMP PREEMPT Mon Dec 11 09:41:01 GMT 2023 aarch64 GNU/Linux
```
```
mks@znp-k1:~$ sudo lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Armbian-unofficial 24.2.0-trunk bookworm
Release:	12
Codename:	bookworm
```

### Additional information

https://github.com/mainsail-crew/mainsail/pull/1021

pedrolamas, 2024-03-05:
> Yes I'm currently running `v0.8.0-319-gf0ae5179`

By any chance are you running a fork of Moonraker? I can't find that version on the official Moonraker repository...

I would expect the suffix `gf0ae5179` to match a commit in https://github.com/Arksine/moonraker, but I can't see any matching

matmen, 2024-03-05:
I think the `g` isn't supposed to be there (it's not a hex character) - I assume it's the latest commit https://github.com/Arksine/moonraker/commit/f0ae5179

pedrolamas, 2024-03-05:
This is indeed a bug, though one could argue that the problem is actually upstream in Moonraker, or even on with the [distro](https://pypi.org/project/distro/) dependency that it then uses to retrieve the distribution values.

Having said that, I too experienced this issue in the past with the BTT CB1 and even [submitted a fix to distro repository](https://github.com/python-distro/distro/pull/353) that was merged and released in version 1.9.0 - at the time this fix was enough for the CB1, but might not be the case with other Armbian releases!

pedrolamas, 2024-03-06:
FWIW, I've just pushed a pull request upstream to the `distro` repository to have this fixed directly there.

pedrolamas, 2024-03-13:
I don't see this getting fixed upstream any time soon, so I've now pushed a fix on our side.

PR #1388 (2024-03-13): fix: Armbian distribution information
If the `release_name` starts with "#", we will then use the `release_id` as correct name - this is a specific fix for Armbian distro but should not be a problem for other distros.

Fixes #1385
closed 2024-03-13 by pedrolamas