fluidd-core/fluidd #1390: weblate translations aren't referenced in changelog

thijstriemstra, 2024-03-14:
### Fluidd Version

1.28

### Browser

Firefox

### Device

Desktop PC

### Operating System

Linux

### What happened

The changelog only mentions direct prs, not weblate prs.

### What did you expect to happen

include prs from weblate

### How to reproduce

-

pedrolamas, 2024-03-15:
Hi @thijstriemstra, thanks for opening this ticket.

Indeed, that has been due to the way we have been merging Weblate pull requests (we squash the PRs before merging).

(FYI @matmen)

I've made a change to Weblate to make it squash all commits per language, keeping the original messages and the authorship as a `Co-authored-by:` footer, and we will start to manually merge these pull-requests (fast-forward via command line).

pedrolamas, 2024-03-17:
Translations for [1.29.0](https://github.com/fluidd-core/fluidd/releases/tag/v1.29.0) have featured in the [CHANGELOG.md](https://github.com/fluidd-core/fluidd/blob/develop/CHANGELOG.md#1290-2024-03-16), so closing this now as complete.

closed 2024-03-17 by pedrolamas