fluidd-core/fluidd #1392: Language pack tr

fatih5228, 2024-03-16:
### Is your feature request related to a problem? Please describe

[tr.zip](https://github.com/fluidd-core/fluidd/files/14625184/tr.zip)
Inside the zip file, there is a tr.yaml file. I have completed the Turkish translation. I would appreciate it if you could add it.

### Describe the solution you'd like

Inside the zip file, there is a tr.yaml file. I have completed the Turkish translation. I would appreciate it if you could add it.
[tr.zip](https://github.com/fluidd-core/fluidd/files/14625191/tr.zip)

closed 2024-03-17 by pedrolamas

fatih5228, 2024-03-18:
Hi. I'm taking your time, but errors in the current translation have been identified and corrected. Thank you in advance for your understanding.

[tr.zip](https://github.com/fluidd-core/fluidd/files/14630202/tr.zip)

pedrolamas, 2024-03-18:
Thank you @fatih5228.

We currently recommend all translation to be done via Weblate (you can find the [TR translations here](https://hosted.weblate.org/projects/fluidd/fluidd/tr/)) as we will then get these integrated in our code more easily, so I've taken the liberty of taking the strings you changes applied those changes myself.

PR #1393 (2024-03-17): feat(i18n-TR): Updates Turkish translations
Adds Turkish translations as provided by @fatih5228

Resolves #1392