fluidd-core/fluidd #1400: 3/5 update broke webcam functionality on older klipper/moonraker software

billkenney, 2024-03-29:
### Fluidd Version

1.28.1

### Browser

Other (Please tell us below)

### Device

Other (Please tell us below)

### Operating System

Other (Please tell us below)

### What happened

I have a Qidi Max 3, which runs on a modified version of klipper and moonraker (circa mid 2022 to mid 2023). I can't update it without losing screen functionality. If I update to a version of fluidd starting with the 3/5 release with the hash 8666162dca6957c5692e879ba79b1ca78f642a23 (building with npm), the webcam no longer appears in my fluidd interface. When I go to cameras, there's a warning icon next to it, and the webcam is disabled when I click on it. I can enable it, and it works for that session, but as soon as I refresh or open a new session, the webcam is gone again. The Timelapse plugin also no longer has the camera selected for Timelapse. This happens whether I'm using a mobile interface, Qidi slicer, Orca slicer, or any browser. 

Based on the git log description, the update was to integrate with the moonraker webcam api (refactor: Moonraker Webcam API (#1386)), but it would be great if you could also allow us who are forced to run older software versions to continue to use the latest versions of your software. 

### What did you expect to happen

I expected the webcam settings to be saved after the session closes. 

### How to reproduce

Update to fluidd version 8666162dca6957c5692e879ba79b1ca78f642a23 or later and older versions of klipper/moonraker are unable to use the webcam or timelapse functionality. 

### Additional information

I've attached screenshots from 066ece3128af69661ff83a7f181a715175f35fcb (which works as expected), and 8666162dca6957c5692e879ba79b1ca78f642a23 (which doesn't work. 
[image]
[image]

pedrolamas, 2024-03-29:
Hi @billkenney thank you for opening this ticket.

We can only ensure support for the official versions of Klipper and Moonraker, not the miriade of modified versions that exist.

Seems to me your options here are to either:

 - replace the modified Klipper and/or Moonraker versions with the official releases
 - NOT update Fluidd and keep using an older version

As a user, I understand this can be frustrating, but as an unpaid open-source developer, I don't feel we have any obligation to maintain compatibility with said forks that we are even unaware of.

closed 2024-03-29 by pedrolamas

billkenney, 2024-03-29:
No worries. I understand you can't cater to everyone, and I appreciate your work and have enjoyed the platform. Just wanted to make you aware of the issue.

pedrolamas, 2024-03-29:
Thank you for understanding @billkenney!

It was my printer, I would consider updating to mainline Klipper and Moonraker, even if that means losing the external screen - that actually happened to me on my Ender-3 V2, the screen was useless after I put Klipper on that printer!

billkenney, 2024-03-29:
My issue with that @pedrolamas (which is dumb I know), is that I bought the printer second hand and the screen was broken, so I had to order a replacement screen, wait 2 weeks for it to arrive from China, and take my printer apart and replace the screen... before I could start printing. Now I feel like I have to use it!