fluidd-core/fluidd #1403: Orca Slicer scarf seams cause incorrect layer count and display in Gcode Preview

tntfriesen1, 2024-03-31:
### Fluidd Version

1.29.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

When the gcode is sliced in Orca Slicer with scarf seams, Gcode Preview counts each step within the scarf seam as a layer.  Current layer and layer count are shown correctly in the "Printer Status" section.  Example, my current print which has 152 layers is showing as having 610 layers in the "Gcode Preview" section.  

Additionally, the drawn lines that show per layer, do not all show for the current layer as they appear to be distributed across the additional "scarf seam layers".

### What did you expect to happen

I expect that the Gcode Preview would show the appropriate number of layers, taking scarf seams into consideration.  I also expect that the lines shown for each layer, match the actual extrusions for that layer.

### How to reproduce

Slice and print any object in Orca Slicer 2.0.0 or above being sure to set the "Scarf join seam" value to select "Contour" and ensure "Scar steps" is some value greater than 1.  In my example, I used 10.

### Additional information

[image]
[image]

pedrolamas, 2024-03-31:
Hi @tntfriesen1, thank you for opening this ticket.

I am going to assume you are not using `SET_PRINT_STATS_INFO` as part of your Slicer configuration, so Fluidd will try and estimate layer times.

To have fully accurate layer counts, you need to use `SET_PRINT_STATS_INFO`.

Inside Orca, please do the following:

- On the top left, click to edit the Printer Preset
  [image]
- On the Printer Settings window, go to the Machine G-code tab
  [image]
- On the Machine start G-code, add this on BEFORE any other command:
  ```text
  SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
  ```
  [image]
- On the Layer change G-code, add this BEFORE any other command:
  ```text
  SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1}
  ```
  [image]
- Save the preset changes

After this, any file sliced will have the layer information added to it and exposed to Klipper (and thus, to Fluidd and other interfaces that use that data)

tntfriesen1, 2024-03-31:
I did not know about this at all.  This is very helpful.  Thank you!

I can confirm that after adding the macros you specified above, the Gcode Preview is correctly showing layer count and proper lines per layer.

Thanks again!

closed 2024-03-31 by tntfriesen1

pedrolamas, 2024-03-31:
Thank you for confirming @tntfriesen1!

I do need to add all this information to the Fluidd docs, but haven't found the time to do it yet...

pmbroth, 2024-08-05:
Thank you - It worked!!

zakdma, 2025-05-02:
@pedrolamas 

This approach does not work. Still need to reload the page to see the layers progress.

kaushikbm, 2025-06-18:
It worked perfectly with my K1. Had this going on for quite sometime.
Only thing is that you have to put the strings before any commands as mentioned.

Thank you @pedrolamas