fluidd-core/fluidd #1419: How to add Authenticator for fuiid.

SlimRG, 2024-04-22:
### Is your feature request related to a problem? Please describe

I have static ip (https://slimhome.tech/#/login) and fluiid.
So, my pass was brootforced by me by 45 days.
It's a secure problem, so I want to fix it.

### Describe the solution you'd like

After entering login/pass show page to write google auth pin or with QR code to create it if no key inserted before.

### Additional information

https://www.rfc-editor.org/rfc/rfc6238
https://github.com/googleapis/google-auth-library-nodejs

pedrolamas, 2024-04-22:
Fluidd runs on the browser, so there is no way to ensure authorization (that would always have to be performed server side).

Such a feature would first need to be implemented as part of Moonraker which provides all the server side APIs to Fluidd.

SlimRG, 2024-04-22:
https://github.com/Arksine/moonraker/issues/842

SlimRG, 2024-04-22:
> If someone somehow bypasses the Moonraker Authentication, they will have full access to your printer, to the point that they could turn the heater on until it melts the whole thing and causes a fire!
> 
> Use a VPN instead to ensure secure access to your resources or something like Cloudflare Tunnels.

If someone bypasses my home assistant - he could open terrariums, so.... But I uses secure two factor system and so...

SlimRG, 2024-04-23:
Add pull request with support for moonraker:
https://github.com/Arksine/moonraker/pull/844

closed 2024-04-24 by SlimRG