fluidd-core/fluidd #1420: Scales and the Image

andora1968, 2024-04-23:
### Fluidd Version

v1.29.1

### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

when printing on the home page
printing percentage and the image of the printed object does not appear

But if I zoom in Chorome 65% in the browser, it appears.

Scale 100%
[image]

Scale 65%
[image]

### What did you expect to happen

At all scales
shouldn't it appear?

### How to reproduce

At all scales
shouldn't it appear?

pedrolamas, 2024-04-23:
Hi @andora1968 thank you for opening this ticket.

This is however not a bug but actually a limitation of the layout space we have available to display all elements.

In this case, if the available width is below a certain threshold, we need to exclude some elements to ensure that the information is still presented correctly, which is the behavior you are experiencing here.

andora1968, 2024-04-23:
> Hi @andora1968 thank you for opening this ticket.
>
> This is however not a bug but actually a limitation of the layout space we have available to display all elements.
>
> In this case, if the available [...]

thanks for your answer