fluidd-core/fluidd #1421: Login in PC version has invisible symbol

SlimRG, 2024-04-23:
### Fluidd Version

v1.29.1-fef6e12

### Browser

Chrome

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Linux, Android

### What happened

When I create account on PC - I can join on PC
But when I connect local mobile and see invisible on PC symblol.

### What did you expect to happen

[image]
And also, when I create account on mobile without this symbol - I can't connect by PC

### How to reproduce

Try to create and login by mobile and by pc without localhost

SlimRG, 2024-04-23:
From PC:
[image]
From Android:
[image]

SlimRG, 2024-04-23:
Upper login was created by phone, another - by PC with MS Edge

pedrolamas, 2024-04-23:
I am unsure of what has happened here... while it looks like you have 2 accounts with the same username, that is not the case as Moonraker would not allow duplicate accounts.

I am unsure of what this user character is, but my advice would be to remove the account and create it again.

SlimRG, 2024-04-23:
I made two accounts to login by phone and by PC

SlimRG, 2024-04-23:
For PC I made on PC and didn't see any invisible symbols.
But when I tried to connect by phone - I can't login with my login (SlimRG)
 So, I created second account through trusted user by phone with no invisible symbols.
 It's not a big problem to have two accounts (for PC and Phone)
 But, it's a bug, that you can reproduse.
 Oh, I forgot to say, PC on Linux (KDE Neon)

SlimRG, 2024-04-24:
As I think, you can trim login string and convert to a-Z0-9@. format

SlimRG, 2024-04-24:
I reinstall fluidd - the same problem

pedrolamas, 2024-04-25:
I have been unable to reproduce this issue with either my Windows PC running MS Edge, or on my Android phone also running MS Edge...

The only reason I see for this thing to happen would be if some browser extension is doing something to change the content of the data submitted!

Can I ask that you access `http://<moonraker-host>:715/access/users/list` and paste the response here?

SlimRG, 2024-04-29:
`{"result": {"users": [{"username": "SlimRG", "source": "moonraker", "created_on": 1713889753.240955}, {"username": "SlimRG\u007f", "source": "moonraker", "created_on": 1713896388.1075256}, {"username": "Test", "source": "moonraker", "created_on": 1714075482.5925968}]}}`

SlimRG, 2024-04-29:
> I have been unable to reproduce this issue with either my Windows PC running MS Edge, or on my Android phone also running MS Edge...
>
> The only reason I see for this thing to happen would be if some [...]

I have extensions LastPass and AdGuard. But on other sites I had no problems with invisible symbol in login. Hmmm...

pedrolamas, 2024-04-29:
The `\u007f` character above is the `Delete` control character... not sure what is going on here, but the browser seems to have entered that in one of the usernames!

pedrolamas, 2024-05-06:
I haven't been able to reproduce this issue. I have only tested in Windows, but my gut feeling is that KDE input is somehow at play here as the `\u007f` character is the `Delete` control character - which should not feature on the input.

Nevertheless, I don't think this is a Fluidd issue and will now close this ticket, but keep an eye on for any other user having similar issues.

closed 2024-05-06 by pedrolamas