fluidd-core/fluidd #1426: Add bulk job delete to web UI

MohnJadden, 2024-04-29:
### Is your feature request related to a problem? Please describe

We have to click on a job -> click on Remove -> click Confirm when we want to delete a gcode file as a job.  Doing this for multiple files is a long, drawn-out process when there are multiple files to delete.   

### Describe the solution you'd like

A column should be added to the Jobs window with a checkbox for each job.  We should be able to check off the checkbox for each job we want to delete, then click a Delete button (similar to how we have Filter and Add buttons) to delete the selections.  The confirmation dialog should reflect that we want to delete X number of selected jobs, where X = the number of checkboxes checked.

### Describe alternatives you've considered

SSHing into the printer and deleting by file name (not practical, since we would have to engage in the same manual process, if not longer) 

SSHing into the printer and finding some way to script deletion (I am not a programmer and wouldn't know where to begin) 

Deleting jobs as we go (not practical if you're sending multiple iterations of a file after each one completes - e.g. testing, batch jobs, etc.)

MohnJadden, 2024-04-29:
Thank you for pointing that out - I wasn't aware. 

Would it be possible to apply this setting to the Jobs menu on the Home page?

pedrolamas, 2024-04-29:
> Would it be possible to apply this setting to the Jobs menu on the Home page?

Short answer is no; the Dashboard on the main page uses a more compact view, mostly for mobile view but also to make it a bit more "light".

pedrolamas, 2024-04-29:
It's not just the width that would get smaller due to the introduction of the checkbox, but also the height of the entries: we currently use the "compact" view of the list on the Dashboard (items has smaller fonts and margins, so each row is actually a bit smaller in height).

Layout is very tricky, especially on the myriad of mobile devices available, so for the moment, there are no plans to introduce the checkboxes in the Dashboard view, and have them only on the Jobs page.

closed 2024-04-30 by pedrolamas