fluidd-core/fluidd #1434: cannot open files

BETLOG, 2024-05-17:
### Fluidd Version

1.30.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

Clicking files in fluidd cannot open them
[image]

### What did you expect to happen

Files should be editable as usual

### How to reproduce

click a file in fluidd interface

### Additional information

This began when a GUI 'update all' was interrupted by running out of disk space.
Config , logs, example configs and gcode are unviewable.
Files are editable via ssh, and a cursory check of permissions looks normal (permissions.txt and all logs are in the zip)
[2024-05-17_fileAccess.zip](https://github.com/fluidd-core/fluidd/files/15345384/2024-05-17_fileAccess.zip)

Even after installing a new, preflashed 32GB EMMC the problem persists. Unfortunately I forgot to check if the file editing issue existed before I did a full update of fluidd/klipper etc, and copied my printer.cfg to the host. So I dont know for sure that it is the update, or my config that breaks it. As far as I remember I havent changed by configs for several weeks, and they look normal to me, so I'm thinking its an update problem.

The printer otherwise prints, and the fluidd GUI functions as normal.
I cannot save endstop adjustments made on the klipperscreen...
but can save zprobe adjustments made on the klipperscreen.

I only got one potentially useful error prior to switching in the new EMMC, but I cannot reproduce it now, despite still not being able to view/edit any file.
[image]

PermissionError: [Errno 13] Permission denied: '/home/mks/printer_data/logs/fluidd-access.log'

pedrolamas, 2024-05-17:
Hi @BETLOG, from your comments and that small error in red, it looks like you have some permission issue.

I can't say for sure what is going on here, but seems the Moonraker service credentials might not have access to the files - whatever this is, it is outside of the scope of Fluidd, as the issue is definitely on Moonraker side.

I did notice that the logs you uploaded do not have the error message shown above (or any other error message) which seems quite odd.

How did you install Klipper, Moonraker and Fluidd? We recommend Kiauh to ensure everything gets correctly installed and configured.

BETLOG, 2024-05-17:
> definitely on Moonraker side.

Noted. Thanks.
https://github.com/mainsail-crew/moonraker-timelapse/issues/163

> How did you install Klipper, Moonraker and Fluidd? We recommend Kiauh to ensure everything gets correctly installed and configured.

> > new, preflashed 32GB EMMC

closed 2024-05-17 by BETLOG