fluidd-core/fluidd #1439: Just improvements

yw4z, 2024-05-29:
### Is your feature request related to a problem? Please describe

Hi,
Previously we discussed some additions on this topic https://github.com/fluidd-core/fluidd/issues/1362

1- I found this one is skipped. what's your opinion on this

### Dashboard > Fans & Outputs >
• Distribute objects to columns by height instead of number. using number as calculation gives not equal spacing for columns. you can see there is dead space on left
[image]

2-This one is integrated but its has an issue. it generates date stamp without leading zeros. Currently it creased file with 2024529 but it not works well without leading zeros. it should be 20240529. Here is scenario 203283 (August 3 2032) listed before 20241214 (December 14 2024)

**File Manager**
• Create zips with Date Stamp 2024-02-12-1315-41315 (best for keeping backups and archiving)

3- Also wondering do you want add slicer logo/colors as theme. Like Orca / Prusa / Bamboo Studio. I can contribute for it

### Describe the solution you'd like

-

pedrolamas, 2024-05-29:
Hi @yw4z, thank you for opening this ticket.

Regarding point 1, there is no easy way to distribute regarding height as that is mostly dynamic (the browser decides what height it needs to show an item), but we do know which ones take more space (the sliders!), so that might be an option...

We will have to investigate this further, but I do agree that the current layout is not great.

Good catch on point 2, I have just fixed that on df59f785b25d7ea90245e7e676aef307745d41cb

Regarding point 3, this is a tricky one as we ALWAYS first contact whoever owns the brand to get their permission to use the logos (and sometimes even to get an SVG of the logo if we can't find one) and that contact MUST be initiated by someone on Fluidd dev team (a bit of "proforma" there, but nevertheless required).

yw4z, 2024-05-29:
@pedrolamas  i have no idea atm about how code side works but it could be solve easier if slider controllers 2x and other controller uses 1x height. and it will solve all small vertical alignment issues on this area

thanks for fixing timestamp format

I guess you can act differently to open source slicers but i get your point. just thought seeing slicer logo and its color on device tab would be nice

pedrolamas, 2024-05-31:
I'm closing this down as a duplicate of #1362 in order to keep tickets a bit clean - point 1 can be followed there and point 3 has now been discussed and explained.

closed 2024-05-31 by pedrolamas