fluidd-core/fluidd #1442: Show remaining length from Spoolman

AndreKR, 2024-05-30:
### Is your feature request related to a problem? Please describe

When the Spoolman integration is enabled, the remaining weight on the spool is shown:

[image]

But when I go to Jobs I only see the length of filament needed:

[image]

This makes sense because the slicer doesn't know the density of the filament, that is only stored in Spoolman.

### Describe the solution you'd like

Spoolman knows the density of the filament:

[image]

So if fluidd would calculate and show the remaining *length* instead/in addition to the remaining weight, it would make it easier to find spools with suitable amounts of filament remaining.

AndreKR, 2024-05-30:
Actually fluidd probably doesn't even have to do the calculation. Spoolman shows the remaining length in its UI:

[image]

So I would assume it's available through the API as well.

matmen, 2024-05-31:
Hi, thanks for your request. Would a simple setting to switch between remaining weight and length suffice or would you prefer to see both values at the same time?

AndreKR, 2024-05-31:
The ability to switch to length would be fine for me, I don't need to see both at the same time. Not sure if that is really simpler, implementation-wise. 🤷

matmen, 2024-06-03:
Thanks. A setting to switch between weight and length units will be included with the next fluidd release :)

PR #1444 (2024-06-03): feat(spoolman): add remaining filament display unit setting
Adds a new spoolman setting to select the display unit (weight (g) or length (m)) of the remaining (and total) filament on a spool:

[image]

**Weight (default)**
[image]
[image]

**Length**
[image]
[image]

---

Closes #1442
closed 2024-06-03 by matmen