fluidd-core/fluidd #1447: Print history has inaccurate or missing “print duration” and “used filament” when using T1

Dycus, 2024-06-10:
### Fluidd Version

v1.30.0

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

On the print history page, all prints have a correct “total duration”.
Prints using T0 have correct print duration and used filament.
But prints that use T1 almost always have 0m 0s for print duration, and 0.0mm (or sometimes nonsense, even negative numbers) for used filament.
On occasion, a T1 print will show a correct total duration, but the filament used is still wrong. (A recent print shows 2h 27m 6s / 2h 34m 30s / 58.2 mm for print duration / total duration / filament used. I promise more than 58mm of filament were used on this 2.5hr print.)

I have a dual-extrusion printer (Replicator 2X). I pretty much only use only one tool in a print. (I haven't tested this behavior with a dual-extrusion print yet.)

I've attached a screenshot and klippy.log.
[image]
[klippy.zip](https://github.com/user-attachments/files/15777139/klippy.zip)

### What did you expect to happen

All prints should show a correct print duration, total duration, and used filament in the history, regardless of tool used.

### How to reproduce

Run a print that uses only T1 (or probably any non-T0 tool) and check the history page once finished.

pedrolamas, 2024-06-11:
Hi @Dycus, thank you for opening this ticket.

Fluidd is just the front-end for the data, maintaining the history is a job for Moonraker (the server side API provider), so I think this might be an issue there.

If doing a Fluidd page refresh (F5) still shows the same incorrect data, my recommendation is that you open this ticket on [Moonraker repo](https://github.com/Arksine/moonraker) and follow up there.

Dycus, 2024-06-11:
Yes, it persists after page refresh, host reboot, etc. 

I'll open a Moonraker issue, thanks.

closed 2024-06-12 by pedrolamas