fluidd-core/fluidd #1453: Default "Camera Url Stream" is not user-friendly for WebRTC (camera-stream).

zhou13, 2024-06-18:
### Is your feature request related to a problem? Please describe

I spend quite a bit of time to get the WebRTC (camera-stream) working. There are two problems:
1. The URLs does not default to `hostname:8080`, but `hostname:80`;
2. The default suffix is `/webcam/?action=stream`, instead of `/webrtc`

### Describe the solution you'd like

I hope the following things becomes the default:
[image]

pedrolamas, 2024-06-18:
Hi @zhou13, thank you for opening this ticket.

For the record, the current defaults we use here are to for the MJPEG Streamer, not the WebRTC streamer.

There is no way we can present any sensible defaults on this as we have no idea what streamer the user is going to use, what host, port, path, etc... It will be up to the end user to specify these details.

I should point out that I can see "fluiddpi" in your screenshot which we have discontinued a long time ago, so we can't even take that as a default.

closed 2024-06-18 by pedrolamas

zhou13, 2024-06-18:
> I should point out that I can see "fluiddpi" in your screenshot which we have discontinued a long time ago, so we can't even take that as a default..

@pedrolamas I just installed a new Raspbian OS with kiauh and I just use `fluiddpi` as a hostname for backward compatibility. Since it is just a hostname and hostname can be anything, I think it is irrelevant for the discussion here.

> For the record, the current defaults we use here are to for the MJPEG Streamer, not the WebRTC streamer.
> There is no way we can present any sensible defaults on this as we have no idea what streamer [...]

As you can see in the screenshot, it clearly says that the selected streamer for the WebRTC is "camera-streamer", not "MJPEG Streamer". It is better to provide reasonable default that works for the selected **WebRTC (camera-streamer)**.

In addition, I found "Camera Url snapshot" is confusing in the context of WebRTC. Is it necessary for WebRTC?

pedrolamas, 2024-06-18:
Apologies, I shouldn't have made assumptions made on the hostname.

Regarding the defaults, I am considering removing them completely in order not to create any confusion going forward.

The screenshot URL is still used if you are using timelapse functionality (and should be set as on mjpeg streamer)

zhou13, 2024-06-18:
Thank you! I agree that completely removing them will be a better defaults.