fluidd-core/fluidd #1454: Z-Offset label leds to setup-confusion and needs another Label added to clarify.

FlailAway, 2024-06-20:
### Fluidd Version

2024 fluidd: v1.19.0-6184c73 

### Browser

Firefox

### Device

Desktop PC

### Operating System

Linux

### What happened

After doing a Mesh Level and setting the Z-offset followed by a "Save" the Z-Offset shows "0.000" when in actual fact that is now the Nozzle Offset, i.e.just touching the bed surface. Many 3D-Printer users have complained about "losing the Z offset" as it is all zeroes, so they set it again. This actually doubles the real Z-offset and the prints are a mess.

Fluidd could do with 2-Labels there, 

"Z-Offset = -0.675"  <-- or whatever it is
and
"Nozzle-Offset = 0.000" <-- nozzle to bed ready to print the first layer at 0.3mm or whatever.

Something to **clearly** indicate that that the nozzle is now at the bed level -- or not if the Z-Offset has not been set yet.

[this printer has a poor reputation largely because of that confusion](https://www.reddit.com/r/ElegooNeptune4/comments/16phz63/lets_do_a_test_regarding_the_famous_z_offset_bug/)

Many other threads on this issue with a lot of printers.

### What did you expect to happen

Not adjust the z-offset again ans extend the actual z-to bed distance.

### How to reproduce

See above as the explanation is self-indicating of the cure.

pedrolamas, 2024-06-21:
Hi @FlailAway, thank you for opening this ticket.

I should first point out that from the ticket information you provided, I see that you are using Fluidd v1.19.0, which is quite old (at time of writing, v1.30.1 is the latest) and we have made significant improvements and fixes around this area since then.

Having said that, the z-offset controls in Fluidd are for the "**live z-offset**" (also called "baby stepping"), so users can adjust the z height while printing.

Though I get your suggestion, Klipper does not have a concept of "nozzle offset" (only "probe offset" and "endstop offset", and even those I don't think are currently exposed by the API), so we would be creating something that would add to the confusion by pointing to something that does not exist.

---

This is the whole logic around this area:

Let's consider you turned on the printer and you have on your config a `stored_z_offset` value.

While printing you decide to change the z-offset using Fluidd "live z-offset" controls, so now the `live_z_offset` has a certain value, so at that specific moment the true offset is the result of `stored_z_offset + live_z_offset`

You then decide you want to save this new offset to the printer config, so you press the save button next to the "live z-offset" controls, thus `stored_z_offset = stored_z_offset + live_z_offset`, and so Klipper makes `live_z_offset = 0` so that the real offset is still correct and the same value as before.

kmitchel, 2024-07-10:
> I should first point out that from the ticket information you provided, I see that you are using Fluidd v1.19.0, which is quite old (at time of writing, v1.30.1 is the latest) and we have made significant improvements and fixes around this area since then.

Elegoo ships the Neptune 4, with significantly older software, and actively discourages users from updating.  As long as a user stays within the bounds of their provided experience, then the printer can perform well.  Step outside, then you better know what your doing, cause your not going to have a good time.

I believe this request could be satisfied by simply displaying the current position_endstop and/or probe z_offset next to the live z offset.  I often find myself opening printer.cfg to find out what the current value is.  No need to invent new nomenclature.

closed 2026-10-12 by pedrolamas